reiniscimurs/DRL-robot-navigation #3: simulation train

wxmshiguang, 2021-11-27:
When I configure the environment variables and run the training
$ cd ~/DRL-robot-navigation/TD3
$ python3 velodyne_td3.py
[image]
I am new to ROS. I wonder if there is a problem with the command.
And What should I type to implement the training example in the picture.
Thank you very much

reiniscimurs, 2021-11-27:
Does the program not run after encountering the "QObject" issue?

I don't have the access to the system right now, but if I remember correctly, this issue does not impede the training process and you can ignore it. Let me know if the training is not happening or the robot is not moving after just simply ignoring the issue. Then I will try to set up the system again, but it might take some time.

wxmshiguang, 2021-11-27:
Yes，The robot is moving normally，but the output of training in following figure does not appear.
[image]
Do I need to input any other instructions？

reiniscimurs, 2021-11-27:
No need to do anything. You just need to wait.

Validation happens only after certain amount of steps. After validation is complete, the "Average reward" will also be output to the terminal.
If you'll go through the velodyne_td3.py file you will notice a variable:
`eval_freq = 5e3  # After how many steps to perform the evaluation`
This is how many steps it will take to perform the validation and subsequently give similar output to the terminal.
You can reduce it to perform validation more often and have the output.

wxmshiguang, 2021-11-29:
Thank you very much for your guidance. I have similar output.

reiniscimurs, 2021-11-29:
That's great to hear. I'll close the issue then.

closed 2021-11-29 by reiniscimurs