reiniscimurs/DRL-robot-navigation #6: time.sleep between unpausing and pausing gazebo in def step

AgentEXPL, 2021-11-30:
https://github.com/reiniscimurs/DRL-robot-navigation/blob/dea37acfc65f702f7fa792787e09602416cf85d4/TD3/velodyne_env.py#L201

I noticed the follwoing processes in def step. 
1. unpausing gazebo, which means the robot starts to move in the gazebo environment.
2. the laser_state and dataOdom are obtained
3. time.sleep(0.1) is called for executing action for a while.
4. pausing gazebo.
5. return state (=np.append(laser_state, toGoal))

This makes me confused, since the retured state is not the state after executing action, but actually the state before executing action. However, I think the returned state needs to be input of policy network which generates the action for next step. Then, laser_state and dataOdom should be obtained after time.sleep(0.1) is called, i.e., step 2 and step 3 needs to be exchanged.

Is my understanding correct?

reiniscimurs, 2021-12-01:
Good spot! Yes, you are correct, it should go "publish twist message->propagate state->get reward->return new state and reward"
The action publishing should happen first like:
`    def step(self, act):
        # Publish the robot action
        vel_cmd = Twist()
        vel_cmd.linear.x = act[0]
        vel_cmd.angular.z = act[1]
        self.vel_pub.publish(vel_cmd)`

And only then unpause the simulator. I must have moved the code around at some point.
Can you confirm that if the placing the velocity publishing command there, does not break anything, and it works fine?

AgentEXPL, 2021-12-02:
After changing the code as follows, it also works.  The code obeys the process of  "publish twist message->propagate state->get reward->return new state and reward". In "propagate state", time.sleep is first called then state value is obtained by rospy.wait_for_message.

----------------------------------------
    def step(self, act):
        target = False

        # Publish the robot action
        vel_cmd = Twist()
        vel_cmd.linear.x = act[0]
        vel_cmd.angular.z = act[1]
        self.vel_pub.publish(vel_cmd)

        rospy.wait_for_service('/gazebo/unpause_physics')
        try:
            self.unpause()
        except (rospy.ServiceException) as e:
            print("/gazebo/unpause_physics service call failed")
        
        time.sleep(0.1)

        data = None
        while data is None:
            try:
                data = rospy.wait_for_message('/r1/front_laser/scan', LaserScan, timeout=0.5)
            except:
                pass
        """
        laser_state = np.array(data.ranges[:])
        v_state = []
        v_state[:] = self.velodyne_data[:]
        laser_state = [v_state]
        done, col, min_laser = self.calculate_observation(data)
        """
        dataOdom = None
        while dataOdom is None:
            try:
                dataOdom = rospy.wait_for_message('/r1/odom', Odometry, timeout=0.5)
            except:
                pass
        
        rospy.wait_for_service('/gazebo/pause_physics')
        try:
            pass
            self.pause()
        except (rospy.ServiceException) as e:
            print("/gazebo/pause_physics service call failed")

        laser_state = np.array(data.ranges[:])
        laser_state[laser_state == inf] = 10
        laser_state = binning(0, laser_state, 20)
        done, col, min_laser = self.calculate_observation(data)

        # Calculate robot heading from odometry data
        self.odomX = dataOdom.pose.pose.position.x
        self.odomY = dataOdom.pose.pose.position.y
        quaternion = Quaternion(
            dataOdom.pose.pose.orientation.w,
            dataOdom.pose.pose.orientation.x,
            dataOdom.pose.pose.orientation.y,
            dataOdom.pose.pose.orientation.z)
        euler = quaternion.to_euler(degrees=False)
        angle = round(euler[2], 4)

        # Calculate distance to the goal from the robot
        Dist = math.sqrt(math.pow(self.odomX - self.goalX, 2) + math.pow(self.odomY - self.goalY, 2))

        # Calculate the angle distance between the robots heading and heading toward the goal
        skewX = self.goalX - self.odomX
        skewY = self.goalY - self.odomY
        dot = skewX * 1 + skewY * 0
[... 103 lines cut ...]
        r3 = lambda x: 1 - x if x < 1 else 0.0
        # rl = 0
        # for r in range(len(laser_state[0])):
        #    rl += r3(laser_state[0][r])
        # reward = 0.8 * r1 + 30 * r2 + act[0]/2 - abs(act[1])/2 - r3(min(laser_state[0]))/2
        reward = act[0] / 2 - abs(act[1]) / 2 - r3(min(laser_state[0])) / 2
        # reward = 30 * r2 + act[0] / 2 - abs(act[1]) / 2  # - r3(min(laser_state[0]))/2
        # reward = 0.8 * r1 + 30 * r2

        self.distOld = Dist

        # Detect if the goal has been reached and give a large positive reward
        if Dist < 0.3:
            target = True
            done = True
            self.distOld = math.sqrt(math.pow(self.odomX - self.goalX, 2) + math.pow(self.odomY - self.goalY, 2))
            reward = 80

        # Detect if ta collision has happened and give a large negative reward
        if col:
            reward = -100

        toGoal = [Dist, beta2, act[0], act[1]]
        state = np.append(laser_state, toGoal)
        return state, reward, done, target

closed 2021-12-02 by AgentEXPL