reiniscimurs/DRL-robot-navigation #8: How to set up sources with an anaconda environment and Noetic branch?

AgentEXPL, 2021-12-03:
Hi, thanks for sharing the Noetic branch. I am testing this branch with an anaconda environment.
First, I opened a terminal using non-conda env. Then complie workspace:
----------
$ cd ~/DRL-robot-navigation/catkin_ws
### Compile
$ catkin_make_isolated
----------(It succeeds!)

Next, I have created an conda env, activated it, and installed some modules as follows.
conda activate py3.6.9
--------------------
pip install torch==1.2.0 -f https://download.pytorch.org/whl/torch_stable.html
pip install pyyaml
pip install rospkg
pip install squaternion
pip install attr
pip install attrs
pip install netifaces defusedxml
------------------------

Finally, I opened a new terminal and set up sources:
---------------------- (This two lines are added to activate conda env and source ROS)
$ conda activate py3.6.9
$ source /opt/ros/noetic/setup.bash
-------------------------
$ export ROS_HOSTNAME=localhost
$ export ROS_MASTER_URI=http://localhost:11311
$ export ROS_PORT_SIM=11311
$ export GAZEBO_RESOURCE_PATH=~/DRL-robot-navigation/catkin_ws/src/multi_robot_scenario/launch
----------------------------------------
$ ########### source ~/.bashrc (This line is deleted because it exits conda env py3.6.9 and turns to conda env base)
$ cd ~/DRL-robot-navigation/catkin_ws
$ source devel_isolated/setup.bash
$ cd ~/DRL-robot-navigation/TD3
$ python3 velodyne_td3.py
-----------------------

The result is as follows, the following error occurs :
------------------------------------------------------------------------------------
Roscore launched!
Unable to register with master node [http://localhost:11311]: master may not be running yet. Will keep trying.
... logging to /home/agent/.ros/log/770251f6-541a-11ec-aeed-e7829104ff20/roslaunch-agent-B365M-POWER-172401.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://localhost:46189/
ros_comm version 1.15.13

SUMMARY
========

PARAMETERS
 * /rosdistro: noetic
 * /rosversion: 1.15.13

NODES

auto-starting new master
process[master]: started with pid [172411]
ROS_MASTER_URI=http://localhost:11311/

setting /run_id to 770251f6-541a-11ec-aeed-e7829104ff20
process[rosout-1]: started with pid [172421]
started core service [/rosout]
Gazebo launched!
... logging to /home/agent/.ros/log/770251f6-541a-11ec-aeed-e7829104ff20/roslaunch-agent-B365M-POWER-172431.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.
[... 83 lines cut ...]
    self.run()
  File "/home/agent/anaconda3/envs/py3.6.9/lib/python3.6/threading.py", line 864, in run
    self._target(*self._args, **self._kwargs)
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/impl/tcpros_pubsub.py", line 185, in robust_connect_subscriber
    conn.receive_loop(receive_cb)
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/impl/tcpros_base.py", line 846, in receive_loop
    self.close()
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/impl/tcpros_base.py", line 858, in close
    self.socket.close()
AttributeError: 'NoneType' object has no attribute 'close'

Exception in thread /r1/front_laser/scan:
Traceback (most recent call last):
  File "/home/agent/anaconda3/envs/py3.6.9/lib/python3.6/threading.py", line 916, in _bootstrap_inner
    self.run()
  File "/home/agent/anaconda3/envs/py3.6.9/lib/python3.6/threading.py", line 864, in run
    self._target(*self._args, **self._kwargs)
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/impl/tcpros_pubsub.py", line 185, in robust_connect_subscriber
    conn.receive_loop(receive_cb)
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/impl/tcpros_base.py", line 846, in receive_loop
    self.close()
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/impl/tcpros_base.py", line 858, in close
    self.socket.close()
AttributeError: 'NoneType' object has no attribute 'close'
---------------------------------------------------------------------

AgentEXPL, 2021-12-03:
I'm sorry for the bad type setting. For simpliciy, the first question is whether the way of setting up sources in anaconda env is correct?

> ## Finally, I opened a new terminal and set up sources:
> ---------------------- (This two lines are added to activate conda env and source ROS)
> $ conda activate py3.6.9
> $ source [...]

AgentEXPL, 2021-12-03:
The second question is that the robot is moving normally in RViz, however, the following result is showing in the terminal. Is there any problems? Or, it goes well?

-------------------------------------------------------------------------
Exception in thread /r1/odom:
Traceback (most recent call last):
  File "/home/agent/anaconda3/envs/py3.6.9/lib/python3.6/threading.py", line 916
, in _bootstrap_inner
    self.run()
  File "/home/agent/anaconda3/envs/py3.6.9/lib/python3.6/threading.py", line 864
, in run
    self._target(*self._args, **self._kwargs)
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/impl/tcpros_pubsub.py", 
line 185, in robust_connect_subscriber
    conn.receive_loop(receive_cb)
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/impl/tcpros_base.py", li
ne 846, in receive_loop
    self.close()
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/impl/tcpros_base.py", li
ne 858, in close
    self.socket.close()
AttributeError: 'NoneType' object has no attribute 'close'

..............................................
Average Reward over 10 Evaluation Episodes, Epoch 1: -71.071544, 0.400000
..............................................
Exception in thread /r1/front_laser/scan:
Traceback (most recent call last):
  File "/home/agent/anaconda3/envs/py3.6.9/lib/python3.6/threading.py", line 916
, in _bootstrap_inner
    self.run()
  File "/home/agent/anaconda3/envs/py3.6.9/lib/python3.6/threading.py", line 864
, in run
    self._target(*self._args, **self._kwargs)
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/impl/tcpros_pubsub.py", 
line 185, in robust_connect_subscriber
    conn.receive_loop(receive_cb)
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/impl/tcpros_base.py", li
ne 846, in receive_loop
    self.close()
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/impl/tcpros_base.py", li
ne 858, in close
    self.socket.close()
AttributeError: 'NoneType' object has no attribute 'close'
-------------------------------------------------------------------------------

reiniscimurs, 2021-12-03:
Sorry, I don't know that much about development in conda, so I can't help you there. I don't think that is something I will be working on. If you manage to figure it out, please let me know, we can create another branch for conda development.

As for the second issue, please take a look at #5, that is the laser issue that I am talking about. I just put the noetic branch up and haven't figured out all the kinks yet.

AgentEXPL, 2021-12-06:
I'm sorry that I found few materials about how to deal with ROS and conda, especially for those which execute ros as subprocesses in python scripts. 

I think the way above is OK to source ROS in conda, since the network is able to be normally trained and evaluated, while the following results can be normally shown. The only problem is the second issue. After deploying the newest code (which is submitted 3 days ago), the second issue still remains. But I think this issue is not a big trouble.

-----------------------------------------------------
Average Reward over 10 Evaluation Episodes, Epoch 1: -71.071544, 0.400000
------------------------------------------------------

By the way, if the defined robot is replaced by turtlebot3, the project works well without the second issue.

closed 2021-12-06 by AgentEXPL

reiniscimurs, 2021-12-06:
> After deploying the newest code (which is submitted 3 days ago)

That seems to be the code from the main branch. Did you try the code from noetic branch (updated 2 days ago)? That one should work but let me know if there is an issue. Also good to know that it all works with turtlebot as well.

AgentEXPL, 2021-12-07:
Sorry for misleading. Noetic branch was actually used. I downloaded the branch with chrome and obtained a zip called "DRL-robot-navigation-Noetic.zip" in 14:45, 2021/12/4 Beijing Time.

To avoid mistakes, I re-downloaded and checked the Noetic branch today. The same issue occurs. (Note that the warning occurs after serveral episodes.)