reiniscimurs/DRL-robot-navigation #10: Roscore not running

reiniscimurs, 2022-01-01:
Average Reward over 10 Evaluation Episodes, Epoch 19: -156.269393, 1.000000
Unable to register with master node [http://localhost:11311]: master may not be running yet. Will keep trying.
Recently, when I trained to about 19-23 epoch, there was a problem that I could not connect to the master node. After this problem, the car kept spinning in the simulation and could not enter the next round. What might be the reason

------------------&nbsp;原始邮件&nbsp;------------------
发件人:                                                                                                                        "reiniscimurs/DRL-robot-navigation"                                                                                    ***@***.***&gt;;
发送时间:&nbsp;2021年12月29日(星期三) 凌晨2:31
***@***.***&gt;;
***@***.******@***.***&gt;;
主题:&nbsp;Re: [reiniscimurs/DRL-robot-navigation] About adding LSTM (Issue #9)

 
Hi,
 
I am not really sure how would I be able to help you with this in any way as you are not giving any specific information about it. I don't think I can help you with purely theoretical extensions of TD3 work.
 
The only thing I can say is that I don't understand why would you use VAE encoding for laser data in the first place. Clearly for dimensionality reduction, but what kind of result are you expecting to come out of it?
 
Second, LSTMs are used for time-series data. The implemented TD3 network in this repo assumes independence in the batched data. Meaning, that when a batch is selected for training, it is a single state-action tuple that it is trained on. There is no information about the previous states in the environment, so I am not sure how you would implement LSTM with this repo without significantly rewriting the code.
 
Please provide more specific implementation descriptions in the future, as it is virtually impossible to answer a question when it is phrased in this manner.
 Good luck with your implementation and extension.
 
—
Reply to this email directly, view it on GitHub, or unsubscribe.
Triage notifications on the go with GitHub Mobile for iOS or Android. 
You are receiving this because you authored the thread.Message ID: ***@***.***&gt;

_Originally posted by @hjj-666 in https://github.com/reiniscimurs/DRL-robot-navigation/issues/9#issuecomment-1003483012_

reiniscimurs, 2022-01-01:
That error would appear if your roscore is not running or the connection with roscore was lost.
You can try going through the steps [here](https://answers.ros.org/question/65182/do-not-know-why-the-master-node-is-not-running/).

This issue is unrelated to the LSTM implementation. Please open a new issue in such cases.

hjj-666, 2022-01-02:
Thank you for your reply. I'll try your solution later

hjj-666, 2022-01-02:
I just checked my ROS_ MASTER_ URI, which looks the same on all command lines, is that when I recently run the program, it is normal at first. After running for a period of time, it will not be connected to the master node

reiniscimurs, 2022-01-02:
This seems like a ROS specific problem. Only thing I can suggest is checking if you are not running out of system resources. You can try reducing the maximum amount of states in the replay buffer and the batch size.

hjj-666, 2022-01-03:
Thank you for your reply. I'll try your suggestion

closed 2022-05-09 by reiniscimurs