reiniscimurs/DRL-robot-navigation #11: install issue

kangwen1011, 2022-01-28:
when i successfully install according to the official instructions,I run `python3 velodyne_td3.py`to train,but I meet the follow issue and gazebo has not started

[image]

reiniscimurs, 2022-01-28:
Hi,

Please provide more information about your setup. Mainly Ubuntu and ROS versions.

kangwen1011, 2022-01-28:
thank you for your prompt reply
ubuntu18.04  ros melodic and pytorch 1.10

reiniscimurs, 2022-01-28:
That should work out-of-the-box then. I don't see any errors in the terminal either. Is this the first time running the repo? Sometimes, in my experience if you are running it for the first time, you just might need to wait for the model to load for 10 ~ 15 minutes. I am not sure of the reason, but I suspect it is downloading some Gazebo models in the background and waiting for them to be loaded. Try running it and leaving it in this state for some time and see if it loads.

kangwen1011, 2022-01-28:
thank you, l will try your suggestion

kangwen1011, 2022-01-28:
l have delete the package and reinstalled,is this step need to wait for 10 ~ 15 minutes?

[image]

reiniscimurs, 2022-01-28:
Sorry for the confusion. You didn't need to change anything, no need to delete and reinstall the repo. This image shows that your ROS cannot find the repo anymore.

Here are the steps:

1. Clone the repository
2. Follow the instructions in the readme
3. Start the training with `python3 velodyne_td3.py` (this is how far it got according to the first image)
4. Wait for some time (let's say 30 min to be sure that all the models have been downloaded in the background)
5. See if the training has started.

kangwen1011, 2022-01-28:
l've got it,thank you very much