reiniscimurs/DRL-robot-navigation #15: python2 or python3 in ros and pytorch

LWw586, 2022-04-18:
In your experiment, ROS melody is python2 and pytorch is python3?

reiniscimurs, 2022-04-18:
If I recall correctly, ROS Melodic will install dependencies for python2 and will run them as such in the background. The pytorch and neural network training stuff is run in python3. So, yes, you are correct.

LWw586, 2022-04-19:
ok,thanks

reiniscimurs, 2022-04-21:
Hi,

It might be a similar issue related to https://github.com/reiniscimurs/DRL-robot-navigation/issues/8#issue-1070390371
Would not hurt to try those steps out.

I do not work it conda and will not be able to check this specific issue, but I could speculate that the issue rises because of calling different packages for different python versions. One thing to check in conda would be if you have python2 installed and if you call `python` in terminal, that python2 is initialized and not python3.

You could do this by opening a terminal and calling `python -V` 

AFAIK this should return `Python 2.7` or similar for it to work correctly.

reiniscimurs, 2022-07-21:
Closing due to inactivity

closed 2022-07-21 by reiniscimurs