reiniscimurs/DRL-robot-navigation #16: simulation problems, modules can not be loaded

Trevor233, 2022-04-26:
Hi Reinis, so sorry to bother you. When i ran python3 velodyne_td3.py, the gazebo can not be activated, and the rviz running was a mess like this
[image]

it seems like the modules of world can not be loaded, and the errors was like this
[image]

Could you please tell me how to fix this, thx!

reiniscimurs, 2022-04-26:
Hi,

This error would appear if you are not loading in TD3.world or TD3-2.world environments. The maps are located in DRL-robot-navigation/catkin_ws/src/multi_robot_scenario/launch/ and specified in https://github.com/reiniscimurs/DRL-robot-navigation/blob/be811a4050dfcb5c800a0a0e4000be81d48cfbc5/catkin_ws/src/multi_robot_scenario/launch/empty_world.launch#L16

It looks like you are launching the program with an empty world, if you would like to do so, you should remove the function that takes care of relocating the boxes (that are not there in the world file) in the reset function here.

I think commenting out this line should suffice https://github.com/reiniscimurs/DRL-robot-navigation/blob/be811a4050dfcb5c800a0a0e4000be81d48cfbc5/TD3/velodyne_env.py#L403 
If not, then remove any line with a reference to the cardboard boxes in the env file

Trevor233, 2022-04-26:
Really appreciate your answering! but when I ran the velodyne_td3.py I didn't change any of your code. So it's so weird that the program didn't load the TD3.world. Could you please tell me how to load the TD3.world? I'm such a rookie at gazebo sorry

Trevor233, 2022-04-26:
sorry it's my bad for not providing these details.
The OS version is Ubuntu 18.04, ROS version is melodic, Python version is Python 2.7.17
Yes I would like to use virtual env and it's so embarrassing that I don't quite know which branch was used...
I just reinstall the program and launch it following your readme.md file 
[image]
It can't load any modules in the TD3.world file yet.

reiniscimurs, 2022-04-26:
I will check the problem in detail once i have a bit more time, but it seems to me it is launching the wrong package here. You can see that is looking into /opt/ros/melodic... for the empty_world.launch file but it should be looking into the repo files. You could check if the repo was installed exactly in the same place and way as noted in the description of this repo (cloned into ~/). If not, then some absolute paths might need to be changed.

In any case, it is not the file in catkin_ws/src/multi_robot_scenario/launch that you should check. Check the one in TD3/assets as that is the one actually used in the repo.

Trevor233, 2022-04-27:
Thanks for your patience. I just reinstalled the repo and I add my package path to the bash.rc file.Then I tried to run the program it became like this.
[image]
It appears that it has loaded the right world but the gazebo still can't be activated and there are still some mistakes.I tried to look out the log file the error mentioned but it's so weird I didn't find it.
[image]

reiniscimurs, 2022-04-28:
Hi,

Seems like the right world file is loading at least. What do you mean by 'gazebo still can't be activated'?
Does your gazebo installation run on its own?

Also, check if you can launch the TD3.world or TD3-2.world on its own.
You can do this by:
`cd ~/DRL-robot-navigation/catkin_ws/src/multi_robot_scenario/launch`
`gazebo TD3-2.world`
Let me know what comes out of that

Also just a simple restart of the computer might help if you haven't done that already

Trevor233, 2022-04-28:
Thank you so much I've check the world file it could be launch. And it turns out that I didn't install a module named 'netifaces' under my working env. The problem  has been solved. Again, really appreciate your kind and patience!
[image]

reiniscimurs, 2022-04-28:
Great that you managed to solve the problem!

No worries, development in ROS is a fickle thing and I was absolutely lost when I started out, so I understand trying to solve these issues very well.

Good luck with the development!

closed 2022-04-28 by reiniscimurs