reiniscimurs/DRL-robot-navigation #17: hello,i have a problem about how to use my training model

xiaozeng-zeng, 2022-04-28:
hello,i am using your code,i think it's concise and readable.
after training i have two model and a result.
[image]

[image]

how could i using the model and the result?
i know the next time i using:python3 velodyne_td3.py .i don't using the model and the result .so it trian again.
thanks a lot

reiniscimurs, 2022-04-28:
Hi,

In line 175 (on melodic/master branch) a network loading function is created that will load network parameters from a file.
 https://github.com/reiniscimurs/DRL-robot-navigation/blob/be811a4050dfcb5c800a0a0e4000be81d48cfbc5/TD3/velodyne_td3.py#L175 a network loading function is created that will load network parameters from a file.
You can use it by simply calling this function right after network creation.

To be exact, find the line:
`network = TD3(state_dim, action_dim, max_action)`
Then insert the following code right after it:
`network.load(file_name, directory="./pytorch_models")`
 

If you want to deploy the network, you will have to create a new network file, and just remove the training functionality. This also means that you will not need critic networks and you should load just the actor parameters.

reiniscimurs, 2022-07-21:
Closing due to inactivity

closed 2022-07-21 by reiniscimurs