reiniscimurs/DRL-robot-navigation #19: Robot keeps circling around

Trevor233, 2022-06-07:
Hi Reinis it's me again. When I run your program I found that after several training the robot just stays put and keeps circling around
/home/trevor233/视频/vokoscreen-2022-06-07_21-45-18.mp4
And the rewards just remains the same
/home/trevor233/图片/2022-06-07 21-48-16屏幕截图.png
Could you tell me where the problem is, thanks.

reiniscimurs, 2022-06-07:
Hi, the links to video and image seem to be from your local directory so I cannot access them.

After how many episodes do you see this behavior? Also, what is the terminal output?

reiniscimurs, 2022-06-07:
If nothing has changed in the code, you can try to change the random seed for the initialization, or just simply run the training again. See if that helps. In this example it seems like the training is just not learning anything significant, so you might just want to restart it.

Trevor233, 2022-06-09:
Hi I just restart the program several times and it still has nothing changed. The agent seems doesn't go to the goal and after several collision it just stay put

Trevor233, 2022-06-10:
Hi thanks for your suggestion. I changed the seed value and the agent seems can finally learn from the training. But sometimes situation like this could happen.It appears that the agent doesn't know to go to the goal. Would you mind telling me how this happened?
[image]

reiniscimurs, 2022-06-10:
In any optimization problem, the starting location of your optimization algorithm and the step size are very important factors to its success. Similar thing is most likely happening here, where the random initialization of network and its weights are located in a local optimum, and any sort of learning with the set learning rate does not allow it to escape it with the current information at hand. So either you need to find a seed value that will initialize the network in a way that works out well, or need to change learning rate, or use some sort of bootstrapping, or change the reward function, or any other method. Essentially, perform hyperparameter tuning. 

For me though, the implementation in this repo has worked fairly well and is usually training fine but that could just be my setup. If you have changed something in the code, that can be something to look at as well and see if that should influence the training.

reiniscimurs, 2022-07-21:
Closing due to inactivity

closed 2022-07-21 by reiniscimurs

vamsi8106, 2023-11-02:
> Closing due to inactivity

Please, May I know the seed value to be set? I have tried many values, but the robot is not converging.