reiniscimurs/DRL-robot-navigation #21: Why using 2 lidar sensors ?

Bab0one, 2022-07-21:
Hello, the compiling of the code works very well, but I just have a question:

Why do we use 2 different lidar (the hokuyo and the velodyne) ?
Wouldn't one lidar be enough for training ?

reiniscimurs, 2022-07-21:
Hi, simply put, the 2d laser is there to detect the collisions while 3d laser is turned into laser state that represents the environment. The 2d laser data is returned by ros topic already as distance information, therefore it is easy to use for collision detection, while 3d velodyne data is returned as point-cloud so requires more calculations to turn it into distance information. Therefore, using the 2d laser for collision detection is more "fool-proof".

In theory, you could detect collisions from 3d data as well and remove the 2d laser completely, and perhaps that is something that I will do when I get around to updating and simplifying the code in this repo.

Bab0one, 2022-07-22:
Ok very good. Thanks for your answer!

closed 2022-07-27 by Bab0one