reiniscimurs/DRL-robot-navigation #22: did you use POI extracting method in this program

Trevor233, 2022-07-26:
Hi, sorry to bother you again, I would like know whether the POI extracting method you mentioned in your paper was used in this program.

Trevor233, 2022-07-26:
I would like to know whether the POI method perform well only for autonomous navigation task. Thanks!

reiniscimurs, 2022-07-26:
Hi,

This specific repo is just about learning robot motion to a given goal. The POI method is not relevant here and is relevant only to exploration tasks as in  https://github.com/reiniscimurs/GDAE and the paper. In this repo, we only learn how to move, not how to choose which point to move to.

POI is just a method to choose a point in the environment to go to next, in order to arrive at the global goal in an unknown environment. In theory, it does not matter what is the agent and how it moves to this point so it might be applicable to something else as well.

Trevor233, 2022-08-03:
ok I get it, really appreciate it!

closed 2022-08-03 by Trevor233