reiniscimurs/DRL-robot-navigation #23: running error

namjiwon1023, 2022-07-27:
Thank you for your contribution!!

When I run the command 'python3 velodyne_td3.py', agent's training episode steps is not displayed.

And the following error will appear.

[image]

How can I solve this problem, waiting for your reply.

reiniscimurs, 2022-07-27:
Hi,

Can you show the full output from the terminal?
Which ROS version are you using?

namjiwon1023, 2022-07-27:
Hi, thanks for replying so quickly！

OS : Mint 19 (Ubuntu 18.04)
ROS version : Melodic

[image]
[image]

After a period of time, information such as episode, mean_reward can be displayed. But I don't know what the impact of the "Spawn service failed" error will be.

reiniscimurs, 2022-07-27:
So as it appears to me the rviz is trying to spawn the robot model but timing out and essentially ignoring the robot model in the end. You can try increasing the time that neural network waits for the environment to start here: https://github.com/reiniscimurs/DRL-robot-navigation/blob/d51438aea0dd8404ae7a633dcfc4f5f96e54f569/TD3/velodyne_td3.py#L210

See if that helps. 
What is the output from the rviz screen?

Also, it seems that it is training just fine, just the robot model is not getting loaded in rviz.
You could check this by opening the gazebo simulator during run time.
Start the training, then open a separate terminal window and run command `gzclient`
This should open gazebo and robot motion should be visible there.

namjiwon1023, 2022-07-28:
I didn't make any changes.

Now the agent has been trained for 20 hours, The agent has learned a policy to find the target.

[image]

rviz screen output:

[image]

I will continue my research in your direction.

I'm very much obliged to you.

closed 2022-07-29 by namjiwon1023