reiniscimurs/DRL-robot-navigation #25: Not found TF frame Odom

yuqikong, 2022-08-14:
Have r1/Odom, but rviz prompt not found odom, Please what should I do to solve this problem?
[image]

reiniscimurs, 2022-08-14:
Hi,

Please provide a little bit more detail.
Mainly, your system setup, ros version, repo version (melodic or noetic) and a full image of the graph.

yuqikong, 2022-08-15:
[image]
[image]
[image]
[image]

yuqikong, 2022-08-15:
I think r1/odom should be changed to odom, but I don't know where to change it.

reiniscimurs, 2022-08-15:
Hi,

I see you are using the Melodic version of the repo on a Kinetic ROS version. There are differences in how things are handled between versions. I cannot be sure that it is happening in this case, but that may cause an issue.

The issue seems to be that there is no link created between r1/odom and base_link. This is explicitly established in https://github.com/reiniscimurs/DRL-robot-navigation/blob/a89e460902e461ee55535180f5ad0d9a5630ae11/catkin_ws/src/multi_robot_scenario/launch/multi_robot_scenario.launch#L19

There should not be a link between r1/odom and r1/base_link as it is in your graph, so make sure all the installation steps were carried out properly (with changed locations, if there were any) and that ros is not calling any other `multi_robot_scenario` package.

You could also explicitly create a link between r1/base_link and base_link and see if that helps.
You can do that by adding 
`<node pkg="tf" type="static_transform_publisher" name="base_to_base_r1"
     args="0.0 -0.0 0.0 0.0 0.0 0.0 base_link r1/base_link 200" />`

right after the previously mentioned code and see if a link is established.
If nothing changes, something might be off with your `static_transform_publisher` node.

yuqikong, 2022-08-15:
I get it, thank you very much.

tqlblzj, 2022-10-20:
> I get it, thank you very much.

i meet the same error,how did you fix it

closed 2022-11-15 by reiniscimurs