reiniscimurs/DRL-robot-navigation #28: 'NoneType' object has no attribute 'close'

blackjacket996, 2022-08-25:
Hey,bro! I've got a issue when I run the velodyne_td3.py. I can see the rviz and gazebo after the command "gzclient", but I could not see the result like epoch, rewards and so on...Do you have met problems like this?
[image]

reiniscimurs, 2022-08-25:
Hi, can you please provide some more information:

- OS version
- ROS version
- Python version
- Using virtual env or not
- Which branch was used (master or noetic)
- Installation steps
- Launching steps of the program

I have seen a similar error when the paths were not set up correctly. I see the path to your repo is different from the one in the description, so make sure to update the absolute paths accordingly.

reiniscimurs, 2022-08-25:
If that does not work, you can try the solution mentioned in https://github.com/reiniscimurs/DRL-robot-navigation/issues/14#issuecomment-1107967435

blackjacket996, 2022-08-25:
> 

I use a server computer which runs Ubuntu18.04 with ROS melodic, python3.6.9. and I follow the steps you mentioned in the first page to run the scripts "velodyne_td3.py".

blackjacket996, 2022-08-25:
> If that does not work, you can try the solution mentioned in [#14 (comment)](https://github.com/reiniscimurs/DRL-robot-navigation/issues/14#issuecomment-1107967435)

OK I'm going to try

blackjacket996, 2022-08-25:
> > If that does not work, you can try the solution mentioned in [#14 (comment)](https://github.com/reiniscimurs/DRL-robot-navigation/issues/14#issuecomment-1107967435)
> 
> OK I'm going to try

Thanks a lot ! I've solved it by using the method mentioned in  #14

closed 2022-08-25 by blackjacket996