reiniscimurs/DRL-robot-navigation #29: navigation in real world

blackjacket996, 2022-08-26:
Hey, Bro! Sorry to bother you again! I wonder if I can use the network model trained here as the path planning module , just make lidar data, distance, angle as the input ,  the network output the velocity which will be published to the move_base . Does it make sense?

reiniscimurs, 2022-08-26:
Hi,
I don't understand what you mean. Please explain the idea in detail, or else I cannot guess what you mean by your proposal.
What is meant by path planning module? Something to replace rrt/a* or a similar path planner?
That would not make sense in this scenario as you do not have map information and the network outputs action directly for a single state observation.

blackjacket996, 2022-08-26:
Sorry, I didn't make it clear. yeah~ as you said , I wanna let it take the place of the local path planner in ROS navigation stack, and before that I will get a 2D-map by a SLAM algorithm such as gmapping, and read the map by map_server or in other ways.

reiniscimurs, 2022-08-26:
Theoretically, you could perform a full path roll-out with the neural network, record the trajectory, and pass that as a planner output. You would require a prior map and a way to extract the laser data from your map information to use this repo directly. I can see some benefit in it as that would allow teaching a deterministic path planner, that could obtain a path faster than let's say an A* approach but it would not be optimal and unreliable as this method is prone to fall into local optimum (as there is only single state input) and that seems to be complicating things quite a bit.

If you would still execute the robot motion with the same neural network, that approach becomes a bit pointless though as the local path planner is not required for execution anyway (refer to paper [here](https://ieeexplore.ieee.org/document/9645287?casa_token=Nyh7Jsg9i20AAAAA:fPKvmgDKl8GIxSRU819t3LL9gyfVfjvOmDXx-yFYdcoCtaKhLmtFtAUd5rPZhop491-f5a6r)). 

The proposed method could be seen in a similar way with a DWA planner as well. Lets say you execute a dwa planner plan in a simulation, record the path and use that as your global path. Sure, you could do that, but why would you?

So could you do it? Probably yes. But you should think what benefit exactly would such an approach bring you.

blackjacket996, 2022-08-28:
Thanks for your reply. I will think it over again to make clear what the purpose is. Sincerely appreciate it！

closed 2022-09-02 by blackjacket996