reiniscimurs/DRL-robot-navigation #30: model [cardboard_box_$$] does not exist

mincheulkim, 2022-09-01:
Hello, thanks to your great work.

When I compiled the code and run the training, the robot runs on only empty space both RViz and gazebo, not in the kind of static obstacle world.

In the terminal, below error message are shown.

[ERROR] [1662014359.011517132, 2.667000000]: Updating ModelState: model [cardboard_box_0] does not exist
[ERROR] [1662014359.011602175, 2.668000000]: Updating ModelState: model [cardboard_box_1] does not exist
[ERROR] [1662014359.011676078, 2.668000000]: Updating ModelState: model [cardboard_box_2] does not exist
[ERROR] [1662014359.011693132, 2.668000000]: Updating ModelState: model [cardboard_box_3] does not exist

I expected some static obstacles such as wall, but it doesn't shown.

reiniscimurs, 2022-09-01:
Hi,

Please check the answer in https://github.com/reiniscimurs/DRL-robot-navigation/issues/16#issue-1215832058

Essentially, check the terminal log for any missing packages and check the that the paths to installation have been set up properly.
If that does not work, let me know the full log output from the terminal

mincheulkim, 2022-09-01:
Hello, I set GAZEBO_RESOURCE_PATH incorrectly. Now the program works fine. thank you!

closed 2022-09-01 by mincheulkim