reiniscimurs/DRL-robot-navigation #32: the goal is not shown in Rviz while training

blackjacket996, 2022-09-12:
Sorry to bother you again , I've succeeded running the velodyne_td3.py and training the navigation strategy, but I found something wrong that the goal which should have been seen in RVIZ did not appear.
[image]
[image]

reiniscimurs, 2022-09-12:
Check if "Marker Topic" does not have different topic names there now. I renamed the "vis_mark_array" names to different topic names in the code, but must have forgotten to save the rviz config.

Essentially, just click on vis_mark_array fields and check for other topics there, see if that helps.

blackjacket996, 2022-09-21:
Thanks. I've solved it by changing the frame of each Marker from "odom" to "r1/odom" ，then select different topics in Rviz. Now it works. After that I trained the strategy but found it still cannot move toward the goal after about 180-200epochs, did the training process has so much randomness？

reiniscimurs, 2022-09-21:
Usually you would be able to see some motion towards the goal after already 30~40 epochs. You can try starting the training again or setting different seeds for different random initialization. If that does not help, there might be some issues in the code.

closed 2022-09-23 by blackjacket996