reiniscimurs/DRL-robot-navigation #33: I got stack at somewhere. Help me please.

Hoggaan, 2022-09-14:
Dear Reiniscimurs, 

Few months back, with your help i reproduced your work. Thanks. Now, i was trying to change for the robot model you used. This time i am trying to use Turtlebot3[Burger or waffle]. I am successful partially as when i run the code, Gazebo loads the world and spawns the model. However, the issue is the training is not starting. look the screenshot below. 

ISSUES:
* Laser is not visible even if i enabled 'True' in the Xacro file.
* The program stops as you can see in the screenshot. 

I guess i do not subscribe some topics. But how?
[image]

reiniscimurs, 2022-09-14:
Hi,

It will be hard to see what is wrong from the output here. Please provide the rqt tree of the robot and the published ros topic list when you run the program.

Can also include the xacro file of the robot to see what is happening.

Hoggaan, 2022-09-15:
* Xarc Files - Two most important: - Can't upload them here but will take vew screenshots from them. 
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

* Useful Screenshots
[image]
![Uploading Screenshot from 2022-09-15 14-13-35.png…
[image]
]
[image]
[image]

Hoggaan, 2022-09-15:
This is your rqt_graph:
[image]

Hoggaan, 2022-09-15:
As i closely watched the robot is circulating in it's position. 
[image]
[image]

reiniscimurs, 2022-09-15:
Hi,
I don't see any robot related topics in the rqt tree. What is your `rostopic list` output? What rostopics are running?

Also, what is the TF_tree in rqt? (When opening rqt go to Plugins->Visualization->TF Tree)

The xacro seems that will not be particularly helpful here. When steps did you take to change the robot model?

Hoggaan, 2022-09-17:
This is the rqt_tf_tree of your project:
[image]
This is the mine: 
[image]

ROSTOPIC LIST
[image]

What steps? answer:

I download tutlebot3 from github. Then i just taken the Turtlebot3_Description packet which i added to my project. Then i made launch files which are typically the same as yours. I edited the files they we pointing to files i wanted for example instead of calling Pioneerxrd.xacro i am calling turtlebot_burger.xacro.

Project is running without an error and gazebo spawns the model into the environment. Even the model is circling in it's position. However, i did not set up publishers and subscribers very well. What is your suggestion?

reiniscimurs, 2022-09-17:
Judging by the TF tree, It seems that the velodyne laser is not attached to your robot base. So it will not move when the robot is moving. That means that every input will be the same, so output will also be the same from the network (which would explain the rotating on the spot motion). See if it is so. A simple way to do so is to add the velodyne pointcloud in rviz (which should already be there, but I do not see the output in your screenshot) and see what comes out visually on the screen.

Hoggaan, 2022-09-18:
Hello thanks. I added the velodyne. But look the picture the two hardware are separate physically. Still the robot is not apple to move. 
[image]
[image]

reiniscimurs, 2022-09-21:
The velodyne laser might appear apart from the robot, depending on how you have described their connection. It might be "floating" above the robot, that should not be an issue. But does the laser move around together with the robot and move around the environment?

Can you share the code, where you attached the velodyne puck?

If attatching the velodyne puck does not work, i will try to implement turtlebot base in the repo (but not earlier than next week)

Hoggaan, 2022-09-21:
<xacro:VLP-16 parent="base_link" name="velodyne" topic="/velodyne_points" hz="10" samples="360" gpu="false" min_angle="-1.57" max_angle="1.57" min_range="0.1">
    <origin xyz="0.125 0 0.25" rpy="0 0 0" />
  </xacro:VLP-16>

I copied this part from your Pioneer Xacro file to my Urdf xacro file. The robot is not moving. I did not get any other help from the internet that is why i am stack for almost a week. This is the intial implement of my research project. My focus is multi-robot task allocation. Please share on your experience what you think one may need to know to do that project.

Hoggaan, 2022-09-22:
Hello. 
I would like to tell you that the robot started moving. The problem was topics. I suspected that the Gazebo was publishing topics different from those of your world. However, i tried several times to make things right in terms of topics. But if anyone of them is not publishing or subscribing the right topic nothing will work. Wonderful!

Thanks for your time.

closed 2022-09-24 by Hoggaan