reiniscimurs/DRL-robot-navigation #35: Odom seems not been published and the map cannot load in RVIZ.

525753936, 2022-11-02:
Thanks for sharing your great work.
Some thing strange happens. There is no error or warning in the terminator when i launch the code "python3 train_velodyne_td3.py"  in ''TD3" dir. 
And I modified the "false" to " true" for the "gui" in empty_world.launch to see if the gazebo works.
It turned out the gazebo was working well.
But the Rviz still cannot load the map. 
Here are my outputs  and  resuts.
I would like to ask that could you please give me some suggestions or advice?
[image]
[image]
[image]

reiniscimurs, 2022-11-02:
Hi

Can you specify what are the displayed errors in the Rviz?
I see you are using ROS Melodic. Did you clone the Melodic version of this repository?

525753936, 2022-11-03:
> Hi
> 
> Can you specify what are the displayed errors in the Rviz? I see you are using ROS Melodic. Did you clone the Melodic version of this repository?

Thanks for your reply. I found the question is the rp version i used was not compatible with my ROS_Version, i checked it out, and after changing the version of the rp version, this problem has been solved and the rviz runs perfectly.
Thank you again for your time and your promising work sincerely.

closed 2022-11-03 by 525753936

peipei518, 2023-11-10:
Hello, I can run normally in the noetic version, but I won't be able to switch to the melodic version, I want to know how to clone the Melodic version of this repository,I am not very familiar with this operation