reiniscimurs/DRL-robot-navigation #36: "Updating ModelState" and "RuntimeError: CUDA error: CUBLAS_STATUS_EXECUTION_FAILED" errors

KhuongDiep911, 2022-11-08:
I am getting an error saying the "cardboard box does not exist" when I run the command "python3 train_velodyne_td3.py". I still haven't found a solution for it. Please help me!

[image]

My PC configuration:
- Ubuntu 20.04
- ROS Noetic
- Pytorch 1.11
- Python 3.8.10

[image]
[image]
[image]

reiniscimurs, 2022-11-08:
Hi,

The cardboard box error comes from the gazebo simulator as it cannot find the cardboard boxes (described [here](https://medium.com/@reinis_86651/deep-reinforcement-learning-in-mobile-robot-navigation-tutorial-part4-environment-7e4bc672f590)) in the environment. This probably happens because the correct world file is not being opened.
Check that you have sourced everything properly and changed your absolute paths, when running the program.
More details on how to do it can be found [here](https://medium.com/@reinis_86651/deep-reinforcement-learning-in-mobile-robot-navigation-tutorial-part1-installation-d62715722303).
Try opening the gazebo simulator and see if it is opening the right world file.

However, this box error will not cause a crash. Is the program crashing?
Can you run it without using cuda as a device?

KhuongDiep911, 2022-11-09:
Hello Reinis Cimurs,

Fortunately, I can fix my problem following your advice about CUDA.
[image]

The first problem is my CUDA, so I uninstalled all my CUDA and pytorch version on my computer. Then I reinstalled it by using two page:
- Pytorch: https://pytorch.org/get-started/locally/
- CUDA Toolkit: https://developer.nvidia.com/cuda-11-6-0-download-archive?target_os=Linux&target_arch=x86_64&Distribution=Ubuntu&target_version=20.04&target_type=deb_local

The second problem which is "cardboard doesn't exist" maybe because I clone your repo to my prefer folder. I exported GAZEBO_RESOURCE_PATH to my folder, but it didn't work. Therefore, I decided to clone your repo to my home folder and it works like a charm. 

I write it here if someone has the same problem as me.

Thank you very for your advice!

reiniscimurs, 2022-11-09:
Great that you managed to get it to work!

Installing in the home directory is probably the easiest way to go to source the repo properly, so that is a very good solution.
It could also be done by replacing the absolute paths in source ( in your case I think it would be:
`export GAZEBO_RESOURCE_PATH=~/tin3080/KHUONG/Project2/repos/DRL-robot-navigation/catkin_ws/src/multi_robot_scenario/launch
source ~/.bashrc` if I have understood your path to the repo correctly.)

KhuongDiep911, 2022-11-09:
Yeah, that's right. 
I try to replace the absolute path and it works too.

Thank you!

closed 2022-11-09 by reiniscimurs