reiniscimurs/DRL-robot-navigation #40: Problems with training and tensorboard visualization.

525753936, 2022-11-21:
Hi, Reinis Cimurs,
Thanks for sharing your amazing work!
 I encountered these problems when I wanna visualize the training process. 
You can see  in the training terminator, AttributeError: 'NoneType' object has no attribute 'close' is called. But the training is still running in Rviz and Gazebo.
In another terminator, I run these command  tensorboard --logdir ./runs
in the dir /DRL-robot-navigation/TD3 . As shown, It was said that No dashboards are active for the current data set. Is there any config that I need to change or modify? 
Could you give me some help for these problems?
Thank you very much.

525753936, 2022-11-21:
[image]

reiniscimurs, 2022-11-21:
Hi,

Which version of the repo are you using and what is your setup?
I see that there is a call to ros Melodic in there. The tensorboard stuff is only available in the Noetic branch.

525753936, 2022-11-23:
Hi，thanks for your reply.
I see, the ros version I use is Melodic.
So I just wait for the training finished,right?
Could you please give me some tips or suggestions for the other problem?

reiniscimurs, 2022-11-23:
Just to clarify, you have ROS melodic on your computer, right?
Did you use the melodic branch from this repository in that case?

It appears to me it is the main branch of this repository that has been installed here and that is a ROS Noetic version. It will not work in ROS Melodic.

If you want it to work in your current setup, I suggest cloning the ros melodic branch. However, that branch is not supported anymore and recent updates (such as tensorboard and some code fixes) are not present there.

Most of all I would suggest moving on to ROS Noetic setup for your computer. That will solve all of these issues.

525753936, 2022-11-23:
Thanks for your reply,
I see.
Have a nice day!

closed 2023-01-12 by reiniscimurs