reiniscimurs/DRL-robot-navigation #45: some question about state setting

namjiwon1023, 2023-01-17:
How make 4-step ( t-n ~ t ) depth image states?

reset function : self.first_state_image = np.stack((x_t, x_t, x_t, x_t), axis=0)

step function : next_state = np.append(x_t1, self.first_state_image[:, :3, :, :], axis=1)

my code is right? 
thank you !!!

reiniscimurs, 2023-01-17:
Hi,

The logic in general seems to be correct, but I am not sure if your axis are correct here. What you would want to have is a single image of shape (batch, channels, width, height). On each step keep last 3 recordings in the channel dimension and add the new recording there.

For the second part, I am not entirely sure what you mean about the output size here? With the setup, you have the parameter sizes look correct. If you mean that the output sizes are different from the paper, then that is because your depthwiise and pointwise layers are defined differently here than in the paper. The implementation of the paper was written with an older TensorFlow so the implementation was quite different, but this should do as well.

In depthwise layer you have missing stride value. By default, in PyTorch, the stride value is 1. That means that your kernel will move only one pixel on each pass step. As such, your output size will be almost the same as input size (80, 64). It comes down to (79, 63) due to your padding settings. In the paper, the stride was set to 2, so the output of the depthwise layer became (1, 10, 40, 32), where 10 is the number of channels.
Afterwards, your pointwise layer seems to be the same (10 channels in, 64 channels out).
But in max_pool2d you are also missing the stride parameter. In PyTorch stride defaults to whatever is the kernel size. Since your kernel is 4, your stride was also 4. However, in paper, stride for max pool was set to 2. So we "max pooled" the (64, 40, 32) image to obtain values with size (64, 20, 16).

So add the `stride=2` parameter to both, self.depthwise and F.max_pool2d layers. Also, figure out the right setting for `padding` parameter.

namjiwon1023, 2023-01-18:
much appreciated !!!
this output size is same as paper !

[image]

reiniscimurs, 2023-01-19:
Great that it works out!

closed 2023-01-19 by reiniscimurs