reiniscimurs/DRL-robot-navigation #53: Problem with rviz

525753936, 2023-04-09:
Hi, thanks for sharing your amazing work!
I got a problem when I am trying to replace the robot  with my own robot model, it seems that my tf coordinates had some problems. Here you can see the tf_graph and the rviz errors.
Could you give me some suggestions to enlighten me to address these problems, since I am new to Ros? 
[image]
[image]

525753936, 2023-04-09:
My Ros Verision is melodic18.04，and I can run your melodic version pioneer3dx robot in both rviz and gazebo.

reiniscimurs, 2023-04-09:
Hi,

Please post the rqt tf tree to see if all the links are connected properly

525753936, 2023-04-09:
Hi,
here is my tf_tree shown.
[image]

525753936, 2023-04-10:
hi I change my code and robot model to the ROS-Noetic version, and i check my tf tree carefully, and it looks like running well. but just a few seconds after launch the environment, the program stuck there. And the robot stopped moving ,i checked the tf tree again, that i found the /odom was vanished, could you gave me some suggestions ?

[image]
[image]
[image]
[image]

closed 2023-11-05 by reiniscimurs