reiniscimurs/DRL-robot-navigation #54: The following errors occur during the training phase.

HzcIrving, 2023-05-09:
`[ERROR] [1683619655.303649057, 4.182000000]: Updating ModelState: model [cardboard_box_0] does not exist                             │[master] killing on exit
[ERROR] [1683619655.304061683, 4.183000000]: Updating ModelState: model [cardboard_box_1] does not exist                             │shutting down processing monitor...
[ERROR] [1683619655.304090828, 4.183000000]: Updating ModelState: model [cardboard_box_2] does not exist                             │... shutting down processing monitor complete
[ERROR] [1683619655.304117199, 4.183000000]: Updating ModelState: model [cardboard_box_3] does not exist  `

HzcIrving, 2023-05-09:
`model  [cardboard_box_0] does not exist`

reiniscimurs, 2023-05-09:
Please make sure to write a full description of your problem, before submitting an issue.

Also, make sure you have set up your sources correctly.
More on that here: https://medium.com/@reinis_86651/deep-reinforcement-learning-in-mobile-robot-navigation-tutorial-part1-installation-d62715722303

HzcIrving, 2023-05-10:
Thanks for your response. I have followed the configuration steps on your side and turned on the verbose mode of gazebo. But when I tried to start training, I got the following error: 
`[Err] [Server.cc:379] Could not open file[./TD3-2.world]`
`[Wrn] [Server.cc:338] Falling back on worlds/empty.world` 

I would appreciate it if you could help me solve this problem. Thank you very much.

reiniscimurs, 2023-05-10:
Hi,

This error appears because your implementation does not find the correct world file to load the training environment. Most likely this is due to not properly set up sources.

You can check this with `echo $GAZEBO_RESOURCE_PATH` and see if that leads to where you have placed the repo.

When executing (and every other part that has  `<PATH_TO_FOLDER>` that you should replace) `export GAZEBO_RESOURCE_PATH=~<PATH_TO_FOLDER>/DRL-robot-navigation/catkin_ws/src/multi_robot_scenario/launch
source ~/.bashrc` it needs to lead to your repo.

markgoli, 2023-05-11:
am getting "Illegal instruction (core dumped)" when i execute 'python3 train_velodyne_td3.py' , what could be the cause or problem?

reiniscimurs, 2023-05-15:
The models alone will not help. The issue is not that the models do not exist, but that the wrong map is loaded.

What is happening is that in the launchfile (https://github.com/reiniscimurs/DRL-robot-navigation/blob/main/catkin_ws/src/multi_robot_scenario/launch/empty_world.launch) the script is trying to start with the specified map. For it to do so, the current repository needs to be added to the paths that ROS will look into. That means that it needs to be sourced properly. Because it cannot find the TD3-2.world map, it defaults to an empty map that does not have any models in it. But the script expects there to be models of cardboard boxes, so it outputs a warning.

First, it looks like you are using the Melodic branch of this repository? Is that correct?

Second, please provide step by step instructions that you executed to clone, source, install this repository. Then I can guide you to what might actually be the problem (images are highly recommended)

closed 2023-11-05 by reiniscimurs