reiniscimurs/DRL-robot-navigation #55: Illegal instruction (core dumped) issue

markgoli, 2023-05-11:
hey reiniscrimurs, I tried to follow the giudelines in the readme file very well, am having ubuntu 20.04.6 with python 3.8.10, I installed pytorch as you stated it, and cloned the tensorboard repository on github, after installing the main dependencies and making exports, when I excute "python3 train_velodyne_td3.py" , I get an "Illegal instruction (core dumped)" what could the problem be? Please help. thank you.

markgoli, 2023-05-11:
And please help me understand more about the tensorboard repository (https://github.com/tensorflow/tensorboard), how is it used?

reiniscimurs, 2023-05-11:
This I am not sure I can help you with. Possibly there is some conflict between some packages or your system. What are the specifications of your system?

You can find more information on tensorboard here: https://pytorch.org/tutorials/recipes/recipes/tensorboard_with_pytorch.html
There you will also find a way to install it. You do not need to clone the repository for it as `pip install tesorboard` should do just fine.

markgoli, 2023-05-12:
My system requirements are, Ubuntu 20.04.6, dual booted with Kali Linux,
500gb hard disk, ros-noetic-full desktop, completely installed, pentium
core CPU / processor. Python 3.8.10, for pytorch, I installed the CPU
version with pip, from the link you shared in the readme file, and for
tensors board, I first installed tensorflow with "pip install tensorflow",
then got 2.25.0 as it's version, then after I cloned the tensorboard
repository, us the link you provided, then placed it in the same directory
with the DDR folder.

markgoli, 2023-05-12:
Thanks for the positive reply too please.

reiniscimurs, 2023-05-12:
You would not need to install tensorflow. We are using PyTorch as the traininng API here, so no tensorflow is required. Tensordboard works well with PyTorch (with limited capabilities) and does not require tensorflow installation. I would recommend to create a new virtual environment and make a clean install of dependencies and then make the repository within this environment.

markgoli, 2023-05-12:
Yeah, thanks. I pray that will work out well

closed 2023-11-05 by reiniscimurs