reiniscimurs/DRL-robot-navigation #56: Scan rotating issue

doyeon-danny, 2023-05-19:
Hi, thank you for sharing your good project.

I am applying to change robot to Jackal and 2D laserscan.
My scan data has been rotating and the robot cannot reach to goal position.
Goal position is in the map's boundary, but laserscan data is going out of the boundary.

As I change robot,laser and code, I guess it is affected by 
1. difference of diff drive between Pioneer and Jackal
    
2. Laserscan data
    I adjusted your velodyne_callback function.

3. Jackal's different IMU which makes laser rotates.

[image]
[image]
Laserscan data is getting far from the main grid. I want to locate it on the right position.

I hope you can help me, Thank you.

reiniscimurs, 2023-05-21:
Hi,

I am not sure if I get the issue here. Is it that the laser scan drifts over time?

In any case, it looks like something is off with your transforms. How did you replace the robot model? What steps did you take for that?

I am not quite getting the point 3. Is the laser scan somehow "tied" to the IMU positioning? What does your rqt_tf_tree look like?

doyeon-danny, 2023-05-24:
Hi,
First of all, I changed "pioneer3dx.gazebo.launch" to "jackal.gazebo.launch".
I adjusted some codes to fit with Jackal model.
Codes are below.
`<launch>

  <arg name="robot_name" default="jackal" />
  

    
    <include file="$(find jackal_gazebo)/launch/spawn_jackal.launch">
      <arg name="x" value="-1.0" />
      <arg name="y" value="5.0" />
      <arg name="z" value="0.01" />
      <arg name="yaw" value="0" />
    </include>

    <include file="$(find jackal_navigation)/launch/include/move_base.launch" />
    

  <node pkg="joint_state_publisher" type="joint_state_publisher" name="joint_state_publisher">
      <param name="publish_frequency" type="double" value="30.0"/>
  </node>

</launch>
`
and this is rqt_tf_graph
[image]

I also changed velodyne_callback function liike below code.
`    def velodyne_callback(self, v):
        self.velodyne_data = np.ones(self.environment_dim) * 10
    
        for i in range(len(v.ranges)):
            dist = v.ranges[i]
            # angle = dist + v.angle_min + v.angle_increment * i
            angle = v.angle_min + v.angle_increment * i
            beta = angle
            # beta = angle - np.pi/2.0

            for j in range(len(self.gaps)):
                # if self.gaps[j][0] <= beta < self.gaps[j][1]:
                #     self.velodyne_data[j] = min(self.velodyne_data[j], dist)
                #     break
                if 0.1 <= dist <= 30.0 and self.gaps[j][0] <= beta < self.gaps[j][1]:
                    self.velodyne_data[j] = min(self.velodyne_data[j], dist)
                    break`

You can ignore number3(about IMU).
I would not fully understand your proejct, so, there would be some problems.

zichengS1999, 2024-01-06:
> Hi, thank you for sharing your good project.
> 
> I am applying to change robot to Jackal and 2D laserscan. My scan data has been rotating and the robot cannot reach to goal position. Goal position is in the map's boundary, but laserscan data is going out of the boundary.
> 
> As I change robot,laser and code, I guess it is affected by
> 
> 1. difference of diff drive between Pioneer and Jackal
> 2. Laserscan data
>    I adjusted your velodyne_callback function.
> 3. Jackal's different IMU which makes laser rotates.
> 
> ![scan-rotate](https://private-user-images.githubusercontent.com/60081287/239471263-e3617d6e-7e5f-4b27-a44a-a242d90b14da.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDQ1MDkyNjksIm5iZiI6MTcwNDUwODk2OSwicGF0aCI6Ii82MDA4MTI4Ny8yMzk0NzEyNjMtZTM2MTdkNmUtN2U1Zi00YjI3LWE0NGEtYTI0MmQ5MGIxNGRhLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMDYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTA2VDAyNDI0OVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWVlNTk5MzVmMGE5NWE4NWU1N2Q3MTNkNmJhMDg1ZTc4OGNlYjkxOWQ5YzM3MTRhODI4ZTdmYmFjOTlmYjkxNTEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.w66ueWzklFGWYcR9wb60zdNlGZf3XISfDGMgKN-AaIY) ![scan-rotate2](https://private-user-images.githubusercontent.com/60081287/239472831-0e407496-7442-4536-95b5-deb353009d4b.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDQ1MDkyNjksIm5iZiI6MTcwNDUwODk2OSwicGF0aCI6Ii82MDA4MTI4Ny8yMzk0NzI4MzEtMGU0MDc0OTYtNzQ0Mi00NTM2LTk1YjUtZGViMzUzMDA5ZDRiLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMDYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTA2VDAyNDI0OVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPThhNmNjNTFhZmY1MzFmZTBmNzVkNDA4MGU1YWIyYzQ3YzA2YjQzY2MwYzRlZDk2NjU1NzE1N2U2NWIyN2VhYjAmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.cSYKJ9speqbR9FGqR4x7g-gJdph5jCkja6BLKyqa050) Laserscan data is getting far from the main grid. I want to locate it on the right position.
> 
> I hope you can help me, Thank you.

Hello, I also have the same issue. Have you managed to resolve it?

closed 2024-03-08 by reiniscimurs