reiniscimurs/DRL-robot-navigation #59: Resume Train

LBJjames23, 2023-06-07:
Hello ,I was wondering if you could please provide guidance on how to resume training from where it left off in case of an interrupted training process ? Any suggestions or best practices on how to achieve this would be greatly appreciated.Thank you so much for your time and expertise.

reiniscimurs, 2023-06-07:
Hi,

You can load the model in training from saved checkpoints using function:
https://github.com/reiniscimurs/DRL-robot-navigation/blob/943186fb7f1890700ce215951e92d5cb92031d14/TD3/train_velodyne_td3.py#LL210C4-L210C4

This is the same as loading the model in testing:
https://github.com/reiniscimurs/DRL-robot-navigation/blob/943186fb7f1890700ce215951e92d5cb92031d14/TD3/test_velodyne_td3.py#L65

Be aware that if you restart training, all the hyperparameters set in:
https://github.com/reiniscimurs/DRL-robot-navigation/blob/main/TD3/train_velodyne_td3.py#L221-L238
will also reset. So the training will start off as 'new training' but the model will have pre-trained weights.
If you want to stop the training and then restart it in the same place, you will need to save these hyperparameters and overwrite them when re-starting the training.

LBJjames23, 2023-06-08:
> Hi,
>
> You can load the model in training from saved checkpoints using function: [...]

Thank you for taking the time to reply. Your help is greatly appreciated!

closed 2023-11-05 by reiniscimurs