reiniscimurs/DRL-robot-navigation #62: change gazebo environment and training

hzxBuaa, 2023-06-16:
Hello, dear Reinis Cimurs,

First of all, thank you for publishing such great work. I want to use this model for our own custom robot as a base.

I simplified the Gazebo environment （modify TD3.world）, keep only the surrounding walls and four cardboard boxes.
But when I'm training, it doesn't converge, the car circles in place. Can I ask you a favour? Why is this？

Thank you so much！

reiniscimurs, 2023-06-19:
Hi,

I would not be able to have any guesses here. There is not enough information here. Please provide what exactly was changed in the code and how you are calling the scripts. Are you also using a different robot model when training?

hzxBuaa, 2023-06-20:
HI，

I just made deletions in the “TD3.world” file.

Obstacles such as tables and stools are deleted, and only the surrounding walls and four random obstacles (four cardboard boxes) are kept.

The rest did not change, nor did the robot model.

reiniscimurs, 2023-06-20:
Can you provide the full terminal output?

hzxBuaa, 2023-06-20:
yes

$ python train_velodyne_td3_.py 
Roscore launched!
... logging to /home/hzx/.ros/log/2559db34-0f4a-11ee-bf50-bb6f994d1be6/roslaunch-hzx-System-Product-Name-1942727.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

Unable to register with master node [http://localhost:11311]: master may not be running yet. Will keep trying.
started roslaunch server http://localhost:40167/
ros_comm version 1.16.0

SUMMARY
========

PARAMETERS
 * /rosdistro: noetic
 * /rosversion: 1.16.0

NODES

auto-starting new master
process[master]: started with pid [1942737]
ROS_MASTER_URI=http://localhost:11311/

setting /run_id to 2559db34-0f4a-11ee-bf50-bb6f994d1be6
process[rosout-1]: started with pid [1942747]
started core service [/rosout]
Gazebo launched!
... logging to /home/hzx/.ros/log/2559db34-0f4a-11ee-bf50-bb6f994d1be6/roslaunch-hzx-System-Product-Name-1942756.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://localhost:35697/

SUMMARY
========

PARAMETERS
 * /joint_state_publisher/publish_frequency: 30.0
 * /robot_description: <?xml version="1....
 * /robot_state_publisher/publish_frequency: 30.0
 * /rosdistro: noetic
 * /rosversion: 1.16.0
 * /use_sim_time: True

NODES
  /
    gazebo (gazebo_ros/gzserver)
    gazebo_gui (gazebo_ros/gzclient)
    joint_state_publisher (joint_state_publisher/joint_state_publisher)
    robot_state_publisher (robot_state_publisher/robot_state_publisher)
    rviz (rviz/rviz)
    urdf_spawner (gazebo_ros/spawn_model)

ROS_MASTER_URI=http://localhost:11311/

process[gazebo-1]: started with pid [1942774]
process[gazebo_gui-2]: started with pid [1942778]
process[urdf_spawner-3]: started with pid [1942784]
process[robot_state_publisher-4]: started with pid [1942785]
process[joint_state_publisher-5]: started with pid [1942786]
process[rviz-6]: started with pid [1942787]
[INFO] [1687252163.146121, 0.000000]: Loading model XML from ros parameter robot_description
[ INFO] [1687252163.149604285]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1687252163.150072344]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
[INFO] [1687252163.151152, 0.000000]: Waiting for service /gazebo/spawn_urdf_model
[ INFO] [1687252163.184770754]: Finished loading Gazebo ROS API Plugin.
[... 67 lines cut ...]
..............................................
Validating
..............................................
Average Reward over 10 Evaluation Episodes, Epoch 6: -252.074390, 0.000000
..............................................
Validating
..............................................
Average Reward over 10 Evaluation Episodes, Epoch 7: -250.687688, 0.000000
..............................................
Validating
..............................................
Average Reward over 10 Evaluation Episodes, Epoch 8: -255.268713, 0.000000
..............................................
Validating
..............................................
Average Reward over 10 Evaluation Episodes, Epoch 9: -251.536725, 0.000000
..............................................
Validating
..............................................
Average Reward over 10 Evaluation Episodes, Epoch 10: -252.257719, 0.000000
..............................................
Validating
..............................................
Average Reward over 10 Evaluation Episodes, Epoch 11: -250.500000, 0.000000
..............................................

hzxBuaa, 2023-06-20:
Average Reward over 10 Evaluation Episodes, Epoch 12: -255.689875, 0.000000
..............................................
Validating
..............................................
Average Reward over 10 Evaluation Episodes, Epoch 13: -256.942230, 0.000000
..............................................
Validating
..............................................
Average Reward over 10 Evaluation Episodes, Epoch 14: -258.425072, 0.000000
..............................................
Validating
..............................................
Average Reward over 10 Evaluation Episodes, Epoch 15: -260.090454, 0.000000
..............................................
Validating
..............................................
Average Reward over 10 Evaluation Episodes, Epoch 16: -256.700221, 0.000000
..............................................
Validating
..............................................
Average Reward over 10 Evaluation Episodes, Epoch 17: -252.317310, 0.000000
..............................................
Validating
..............................................
Average Reward over 10 Evaluation Episodes, Epoch 18: -254.078768, 0.000000
..............................................
Validating
..............................................
Average Reward over 10 Evaluation Episodes, Epoch 19: -252.697328, 0.000000
..............................................
Validating
..............................................
Average Reward over 10 Evaluation Episodes, Epoch 20: -254.145063, 0.000000
..............................................
Validating
..............................................
Average Reward over 10 Evaluation Episodes, Epoch 21: -259.087948, 0.000000
..............................................
Validating
..............................................
Average Reward over 10 Evaluation Episodes, Epoch 22: -253.587934, 0.000000
..............................................
Validating
..............................................
Average Reward over 10 Evaluation Episodes, Epoch 23: -252.160117, 0.000000
..............................................
Validating
..............................................
Average Reward over 10 Evaluation Episodes, Epoch 24: -252.280001, 0.000000
..............................................

reiniscimurs, 2023-06-21:
I do not see any errors in the output log here so the map loading should be fine. Do you see your changes in the gazebo simulator and does it appear the way you would expect?

One thing to try is to change the seed and train network with other random initialization.
You can do that here: https://github.com/reiniscimurs/DRL-robot-navigation/blob/943186fb7f1890700ce215951e92d5cb92031d14/TD3/train_velodyne_td3.py#L221

hzxBuaa, 2023-06-25:
ok, thank you very much!

closed 2023-06-25 by hzxBuaa

slamlixiaodong, 2024-10-23:
> ok, thank you very much!

Do you train successfully? I encounter the same issue and change the seed, but it seems to be not useful. If you have done it, could you give me some advice?