reiniscimurs/DRL-robot-navigation #63: baseline-PP

hzxBuaa, 2023-06-19:
Hello, it is an excellent repo for navigation. I am wondering in your paper( https://arxiv.org/abs/2103.07119) you mentioned baseline algorithm- PP. I have two questions.
1. The path planning part uses the Dijkstra algorithm, but what control algorithm is used by the control layer to make the vehicle track the path?
2. Is the travel time mentioned in the paper the time it takes for the vehicle to track the path planned by Dijkstra algorithm？
I hope to receive your reply. Thank you very much.

hzxBuaa, 2023-06-19:
But the Dijkstra algorithm needs a proper plan before exploring,  so is it necessary to obtain all map information to adopt this algorithm?

reiniscimurs, 2023-06-19:
That's not entirely so. You can obtain an initial naive plan without exploration. A grid map has known free, known occupied, and unknown elements. By using move_base we can allow planning through unknown parts of the map, so an initial plan will be obtained. Then when executing it, the unknown parts will become known, and the original plan will be re-planned. Since that takes time, we used heuristics instead and show that through the time that it takes to get to the final goal.

hzxBuaa, 2023-06-21:
Thank you very much for your answer. I think I should understand what you mean. You are using the Dijkstra planning algorithm, and the control layer comes with move_base and ROSARIA. Without the need to build a map in advance, when the car is driving, the lidar senses the surrounding environment, the unknown becomes known, and then the algorithm adjusts the planned path according to the established environment, is this true?

closed 2023-06-25 by hzxBuaa