reiniscimurs/DRL-robot-navigation #66: Robot goes around in circles.

Talha771, 2023-07-01:
I cloned this repo and ran the training, however, there has been no noticable improvement in the reward and the robot just goes around rotating, and does not move at all. Am I missing setting some parameters or getting overly stressed way too early in the training process ? 

[image]

It is moving a bit, but mainly just in circles, Also, I'm running this on an Titan RTX and it trained around 52 epochs in around 15 hours, I think its taking longer than it should, but I don't know how to change that.

reiniscimurs, 2023-07-01:
Hi,

It could just e that you need to change the seed for the initialization of the network.
See explanation here: https://github.com/reiniscimurs/DRL-robot-navigation/issues/19
Let me know if that does not work, then we can look into it deeper.

Talha771, 2023-07-01:
Thank you, will try that. Also thank you for being so responsive

Talha771, 2023-07-01:
This worked perfectly.

DanDan996, 2023-07-08:
> This worked perfectly.

Hi, can you please tell me what your seed value is set at? I have set many values without good results and end up in the situation described in your question.

vamsi8106, 2023-11-02:
> This worked perfectly.

Please may I know which value of seed has worked good I have tried many values but no improvement.

closed 2023-11-05 by reiniscimurs