reiniscimurs/DRL-robot-navigation #67: car model

hzxBuaa, 2023-07-04:
Hi ，Reinis Cimurs，I would like to ask which model the car adopts，ackerman steering or something else？

reiniscimurs, 2023-07-04:
This repo implements a differential drive robot.

hzxBuaa, 2023-07-04:
Ok, thank you. Is this the topic "cmd_vel" directly posted to the car to make it move? Do you need to consider dismantling this command according to the current vehicle kinematics and converting it into the speed and angular velocity commands of each joint? Similar to the conversion of control commands into chassis commands, if this is to be applied to a real vehicle, does the simulation need this step? Or just release cmd_vel directly to let the car move? I don't know if I made it clear, thank you in advance

reiniscimurs, 2023-07-04:
We publish the command to topic "cmd_vel". From the robots side there is a differential drive plugin that then takes this command and uses it for robot controller: http://wiki.ros.org/diff_drive_controller

If you want to use a different kinematic model, you would have to use a different plugin for your model in training.
For instance: http://wiki.ros.org/ackermann_steering_controller?distro=noetic

hzxBuaa, 2023-07-05:
Alright, thank you very much
Finally, I would like to ask. I see that our repo does not use the 'controller_manager' pkg, but I see that my Ackerman differential robot uses this package. I would like to ask why we do not use it? And if this package is not used, will it affect the final effect of transplanting to the real car?
I am a novice, hope you can give an answer, thank you very much!

reiniscimurs, 2023-07-05:
That is outside the scope of this repo and my experience so I cannot help you there. You would have to try it out and see for yourself.

closed 2023-07-13 by hzxBuaa