reiniscimurs/DRL-robot-navigation #68: How to solve: The program does not converge and the collision rate is always 1

DanDan996, 2023-07-07:
[image]
[image]
After I looked at many questions and your answers, I changed the **seed** values of the code (**0, 1, 2, 3, etc.**), but invariably I ended up with training results similar to this:

**Average Reward over 10 Evaluation Episodes, Epoch 48: -106.461269, 1.000000**

And at this point the carriage will perform **[image]**s, even if its position changes and it continues to perform circular motions. This phenomenon does not occur at the very beginning, in the first few epochs we see that the vehicle's strategy is at least normal although there are many collisions, but the later epochs show wrong training results. I don't know how to modify the seed value, are there any other places that I didn't notice?

reiniscimurs, 2023-07-07:
Hi,

It is very hard to say what the issue is here without more information. Please provide full log output of the terminal as well as your system setup and maybe an rviz screenshot.

Additionally, try to take a look at the gazebo simulation and if there are any issues there.

DanDan996, 2023-07-08:
Glad to see your reply, I now want to run the program again. And will give new results and what I have set up later. This will take some time, please wait for my reply.

> Hi,
> 
> It is very hard to say what the issue is here without more information. Please provide full log output of the terminal as well as your system setup and maybe an rviz screenshot.
> 
> Additionally, try to take a look at the gazebo simulation and if there are any issues there.

closed 2023-11-05 by reiniscimurs