reiniscimurs/DRL-robot-navigation #69: Gazebo Not publish TF frame /odom and /cmd/vel

hzxBuaa, 2023-07-12:
Hi, excuse me, after I changed the robot model, it can’t run normally. I checked all the nodes and topics, and found that node "/gazebo" can’t send "/odom"/ and "/cmd_vel" topics like your original one. How can I solve this question? 
Thank you in advance!

hzxBuaa, 2023-07-12:
Also, I see that you included 'base_link' and 'r1/base_link', what is the difference between these two? In rviz, after I changed the Fixed Frame to 'r1/base_link', I found that the car does not move, but the scene moves, but You can also train normally, no difference, right?

reiniscimurs, 2023-07-13:
Please provide full log outputs, Node Graph and TF Tree from `rqt` of your setup.

closed 2023-07-13 by hzxBuaa

Ethan0207, 2024-01-26:
Hello,
I have met the same problem. I checked all the nodes and topics, and found that node "/gazebo" can’t send "/odom"/ and "/cmd_vel" topics. Could you tell me how you solved the problem? Thank you