reiniscimurs/DRL-robot-navigation #70: /scan

hzxBuaa, 2023-07-13:
data = None
    while data is None:
        try:
            data = rospy.wait_for_message('/r1/front_laser/scan', LaserScan, timeout=0.5)
        except:
            pass

    **laser_state = np.array(data.ranges[:])**
    v_state = []
    v_state[:] = self.velodyne_data[:]
    laser_state = [v_state]

Hi, your code radar takes 3D (velodyne_data) 20 sets of point cloud data as the state input, I want to change it into a 2D lidar input, but my 2D lidar here has 720 sets of data, how can I convert the 720 sets Become 20 groups? How to choose?

hzxBuaa, 2023-07-13:
Also, how should I modify the code to change from 3D 'Pointcloud2' to 2D 'LaserScan' in order to modify less code? 
Thank you in advance!

reiniscimurs, 2023-07-13:
I see you are using the melodic branch. Since it is not updated there are some issues there. For instance 
`data = None
while data is None:
    try:
        data = rospy.wait_for_message('/r1/front_laser/scan', LaserScan, timeout=0.5)
    except:
        pass

**laser_state = np.array(data.ranges[:])**`
Is actually never used as state representation. I do think we detect collisions from there.

Please read the tutorial to familiarize yourself with the code. The gaps and laser selection is explained there: https://medium.com/@reinis_86651/deep-reinforcement-learning-in-mobile-robot-navigation-tutorial-part4-environment-7e4bc672f590

You could use it for 2D laser as a source if you want and pass it through the same function if it is set up properly.

Also, you can take a look at another repo where two 2D laser scans are used and 720 laser scan values brought down to 20 representative values: https://github.com/reiniscimurs/GDAE/blob/main/Code/GDAM_env.py

hzxBuaa, 2023-07-14:
laser_state = np.array(
            [min(laser_in[0:20]), min(laser_in[21:60]),
             min(laser_in[61:100]), min(laser_in[101:140]),
             min(laser_in[141:180]), min(laser_in[181:220]),
             min(laser_in[221:260]), min(laser_in[260:300]),
             min(laser_in[301:340]), min(laser_in[341:380]),
             min(laser_in[381:420]), min(laser_in[421:460]),
             min(laser_in[461:500]), min(laser_in[501:540]),
             min(laser_in[541:580]), min(laser_in[581:620]),
             min(laser_in[621:660]), min(laser_in[661:700]),
             min(laser_in[701:719])])
        laser_state = laser_state / 10
Hi, I see the code ,but why 720 laser scan values brought down to 19 representative values?How should one choose？

closed 2023-07-16 by hzxBuaa