reiniscimurs/DRL-robot-navigation #72: Problems with the training process

yuyuanyaya, 2023-07-28:
Hi , thank you for your great project. I had some problems when trying to reproduce your work. I followed the readme documentation to configure the environment and dependencies.

Here is my system setup:
Ubuntu 20.04 LTS
ROS noetic
python 3.8.10
pytorch 2.0.1+cpu
tensorboard 2.13.0
codes are cloned from main branch
Here's a picture of my (.bashrc)
[image]
[image]
This is my cmake operation
[image]
[image]
This is my train operation
[image]
[image]
[image]
At this point there is a problem, and gazebo won't start, can you help me?

reiniscimurs, 2023-07-28:
Hi, Thank you very much for the detailed logs and description. It is very much appreciated. Are you running the repo in a clean install of ROS? If that is the case, it might be that ROS is downloading some models in the background and not able to start the training without them. In such a case, I suggest letting the training "run" for some 20 minutes and then restarting to see if the models are downloaded.

Other option is that odometer data is read before the robot has been spawned. You can try increasing the time to wait for all the things to have spawned in the gazebo.
https://github.com/reiniscimurs/DRL-robot-navigation/blob/943186fb7f1890700ce215951e92d5cb92031d14/TD3/train_velodyne_td3.py#L253

yuyuanyaya, 2023-07-29:
Hi，thank you very much for the suggestion, now it is able to run in rviz. But it is not
[image]
 to display the gazebo environment and the camera image in rviz is also grayed out and no image is displayed. How can I solve this problem?

reiniscimurs, 2023-07-30:
Did you try launching gazebo? (By default it will not be launched automatically). If so, what is the error that comes up?

wzg98, 2023-09-01:
> 您好，非常感谢您的建议，现在已经可以在rviz中运行了。但并不是 显示gazebo环境，rviz中的摄像头图像也是灰显的，没有图像显示。我怎么解决这个问题？ [image]
I have encountered the same issue with rviz as you. How did you solve it? Thank you very much.

closed 2024-03-08 by reiniscimurs