reiniscimurs/DRL-robot-navigation #73: problems when run “python3 train_velodyne_td3.py”

eecai2004, 2023-08-11:
hello！thank you for your wonderful work. I try the melodic version. there are some problems and errors when I run "python3 train_velodyne_td3.py”
before I run the command， I wrote these infos showed below into the .bashrc file 
[image]
then open a terminal and run as images showed below.
1. unable to register with master node
[image]
[image]
2. Error: package 'gazebo_plugins' not found
[image]
[image]
[image]
Look forward to your response and thank you so much.

reiniscimurs, 2023-08-11:
Hi, 
1. That is perfectly fine and it registers with master on the second try.
2. gazebo_plugins should be part of your ROS (with gazebo) installation. Have you installed the full version of ROS including Gazebo?

eecai2004, 2023-08-25:
Thank you so much for your response so quickly!
According to your advice , i found when i run it in a conda environment , it can not find the gazebo-plugins ,then i tried to setup by using the command "conda install -c conda-forge gazebo". then there was a gazebo11(probablys is not a full version) in the conda environment and a gazebo9 outside the conda env ( gzserver: /build/ogre-1.9-B6QkmW/ogre-1.9-1.9.0+dfsg1/OgreMain/src/OgreRoot.cpp:105: static Ogre::Root& Ogre::Root::getSingleton(): Assertion `msSingleton' failed.).  Maybe it's some kind of confusion between versions that caused Gazebo to fail to start, whether in the conda environment or outside of it. After multiple attempts to uninstall and reinstall Gazebo, it still could not be started properly. In the end, I had to reinstall the Ubuntu18.04. Finally I can run it out of the conda env(I don't install conda yet) after wasting more than a week. 
But now there are some new problems looking forward to your help.
1\ when I run the "python3 velodyne_td3.py" in melodic version  ,there is an error showed in the image below. what should i do to fix it?
[image]
[image]
2\  can I replace the .py files in TD3 of melodic version with the .py files in TD3 of Noetic version? Then in order to run it in melodic, do I need to modify any files accordingly?
3\  one more question, in this project ,only lidar data was used to detect obstacles ,have you ever consider multisensor confusion ,for example , confusion of lidar data and camera data ?
Look forward to your response and thank you so much.

reiniscimurs, 2023-08-26:
For answers to 1, see if solution here helps: https://github.com/reiniscimurs/DRL-robot-navigation/issues/28

2. I think if you adapt the ROS topic names, you could just take python files from noetic version and use it in melodic. Howver, I do not recall the exact changes between the two anymore so I cannot guarantee it would work straight away. But mostly, the changes between noetic and melodic version are within the support files and not .py files.
3. Sensor fusion is a topic of its own and more difficult than just adding extra sensors. To obtain proper sensor fusion you would require larger networks or sensor fusion algorithms. The reasons why we use lidar is essentially the ease of use and small number of parameters for navigation. As soon as you start using images, this will blow up your neural network, make it more difficult to train and will be more difficult to deploy to low cost robots. You can take a look at some previous work I have done with CNNs and DDPG for robot navigation: https://www.mdpi.com/2079-9292/9/3/411 The neural network ended up being quite huge and difficult to deploy, not to mention field of view issues. But, in general, sensor fusion could help with navigation policies and scene understanding, especially for some goal or human recognition. It is an interesting topic that could yield some interesting results.

closed 2023-11-05 by reiniscimurs