reiniscimurs/DRL-robot-navigation #76: Whether to support changing to another algorithm

pzhabc, 2023-10-08:
Hi. Thank you for everything. This is great. I would like to ask if I can change the TD3 algorithm to other algorithms based on your work, such as DQN, DDQN and so on. I don't know much about this. I hope you can spare some time to answer my questions. Thanks again!

reiniscimurs, 2023-10-08:
Hi
Yes you could change the algorithm to other DRL algorithms and train the network with your own methods. You could specify a different algorithm class to be called in line:
https://github.com/reiniscimurs/DRL-robot-navigation/blob/101ed8eaba77b9f447e3d57cc71518cfedb7f65c/TD3/train_velodyne_td3.py#L261
This would call a different DRL training, assuming that the class methods (such as `train()` and `get_action()`) are implemented.
It should be relatively painless to adapt a DQN or DDQN algorithm to work here.

pzhabc, 2023-10-09:
I'm sorry to ask you again. Based on your work, I used DQN to implement it. The number of action Spaces of the robot is 5, and the corresponding linear and angular velocities of the 5 actions are shown in the picture below. The neural network outputs the value of the 5 actions, and then the robot chooses one of the actions to execute. However, I encountered some problems during operation, as shown in the picture, after entering the train(), the values of max_value and Q_value are very large, I don't know what caused it, can you give me some help? Looking forward to your reply, thank you!
[image]
[image]
[image]
[image]
[image]

pzhabc, 2023-10-09:
> 嗨，是的，您可以将算法更改为其他 DRL 算法并使用您自己的方法训练网络。您可以指定要在行中调用的其他算法类：
> 
> https://github.com/reiniscimurs/DRL-robot-navigation/blob/101ed8eaba77b9f447e3d57cc71518cfedb7f65c/TD3/train_velodyne_td3.py#L261
> 
> 
> 这将调用不同的 DRL 训练，假设实现了类方法（例如 和 ）。
> 在这里调整 DQN 或 DDQN 算法应该相对轻松。`train()``get_action()`

I'm sorry to ask you again. Based on your work, I used DQN to implement it. The number of action Spaces of the robot is 5, and the corresponding linear and angular velocities of the 5 actions are shown in the picture below. The neural network outputs the value of the 5 actions, and then the robot chooses one of the actions to execute. However, I encountered some problems during operation, as shown in the picture, after entering the train, the values of max_value and Q_value are very large, I don't know what caused it, can you give me some help? Looking forward to your reply, thank you!
[image]
[image]
[image]
[image]
[image]

reiniscimurs, 2023-10-09:
Is the training running alright and at what stage of training do you see these values? How do you come to conclusion that the Q value is too high? If it is a random initialization the Q values would be rather unstable and could be overestimated in the beginning of the training.

pzhabc, 2023-10-09:
> Is the training running alright and at what stage of training do you see these values? How do you come to conclusion that the Q value is too high? If it is a random initialization the Q values would be rather unstable and could be overestimated in the beginning of the training.

Thanks for your reply! The training did not go well, and it was the first step into the train() function that led to this result.

reiniscimurs, 2023-10-09:
Can you share the full code?

It seems a bit weird to me that action is reshaped to shape (-1,1) even though it is a 2 dimensional vector in the regular replay buffer. Then this is used in max_q calculation with gather function. If you are using the DQN from pytorch tutorial, then there it uses a 1D vector for action space so there might be a discrepancy there.

pzhabc, 2023-10-09:
> Can you share the full code?
>
> It seems a bit weird to me that action is reshaped to shape (-1,1) even though it is a 2 dimensional vector in the regular replay buffer. Then this is used in max_q [...]

Thanks for your reply! I mainly changed the actions of the robot in the file "train_velodyne_td3.py", but did not change other files. Please look at the picture.
[image]
[image]
[image]
[image]
[image]

reiniscimurs, 2023-10-09:
Are you implementing the DQN from pytorch (https://pytorch.org/tutorials/intermediate/reinforcement_q_learning.html)?
If not, let me know the source of your implementation.

I do not see any major issues here, but the train function might have some issues here. Not sure if you want to reset the target critic model every second training iteration (line 114). Might be better to reset after the training loop has completed. But generally I would expect a soft update of the target network parameters, which I do not see in this code. Here, the target network just takes over all the network weights as they are on every second weight update. This should make the network quite unstable and learning might fail.

But also did you run it only for one training episode? Do the Q values not reduce over the training? Here we are initializing the neural network with random values so the values being wildly off in the beginning would not be unexpected.

pzhabc, 2023-10-10:
> Are you implementing the DQN from pytorch (https://pytorch.org/tutorials/intermediate/reinforcement_q_learning.html)? If not, let me know the source of your implementation.
>
> I do not see any major [...]

Thank you very much for your help! I referred to the link you provided. The soft update you mentioned was used. After running several training sessions, you find the results as shown below. Does this mean that the training is effective?
[image]
[image]
[image]

But I'd like to ask you a few more questions.

1. "Exception sending a multicast message:Network is unreachable" appears on the terminal. Do I need to be connected to the network for this training?

2. The implementation of DQN algorithm I refer to is not the official website, it does not use soft update, but the official use. Here soft updates seem to play a big role. Is it necessary in the algorithm?

3. For some existing algorithms, can I also refer to the official website provided by you for implementation?

4. May I ask what caused the initial increase in the loss function?

Thank you, you have been a great help to me!

reiniscimurs, 2023-10-10:
1. I am not familiar with this error. Please post the error log.
2.  Here is a good answer to this question: https://ai.stackexchange.com/questions/21485/how-and-when-should-we-update-the-q-target-in-deep-q-learning
While not strictly required, the target network update scheme must make sense.
3. I am not sure what the question is here. Feel free to refer to official websites of implementation if it makes sense.
4. Loss function is the target Q value difference from Q value. Initially you will see smaller target Q values as reward will be low since most likely would not have encountered that many terminal episodes. Then target Q values will be lower and very close to Q values. As such, your loss will increase and then reduce once actual training begins.

It seems to me that some training is happening as average Q value increases. Not sure if you want to cap the max Q value though, but that is up to your design. However, you should be able to visually see how the policy works in the environment. 30 episodes are a rather small amount for any reasonable training though.

pzhabc, 2023-10-11:
Ok, thank you, what you said is quite right. Before, there was too little training, but with the increase of training, I found a problem. Is it because DQN is not suitable for such a scenario or the code itself has defects and needs to be improved?
[image]

reiniscimurs, 2023-10-12:
DQN and DDQN should not have any major issues in learning with this code as there exist successful implementations. I do not see any reason why it shouldn't work from theoretical standpoint. Most likely there is some discrepancy in the code itself.

pzhabc, 2023-10-13:
Okay, thank you. I recently in learning reinforcement learning in this website "https://spinningup.openai.com/en/latest/spinningup/rl_intro3.html" saw such a content: In Policy Learning, we should care about average returns. The loss function means nothing. I wonder if the same is true in value learning?

pzhabc, 2023-10-13:
> DQN and DDQN should not have any major issues in learning with this code as there exist successful implementations. I do not see any reason why it shouldn't work from theoretical standpoint. Most likely there is some discrepancy in the code itself.

Okay, thank you. I recently in learning reinforcement learning in this website "https://spinningup.openai.com/en/latest/spinningup/rl_intro3.html" saw such a content: In Policy Learning, we should care about average returns. The loss function means nothing. I wonder if the same is true in value learning?
[image]

reiniscimurs, 2023-10-13:
What they mean here is that in typical imitation learning tasks you would have ground truth and your calculated observation. Loss is expressed as the difference between ground truth and the observation. Therefore, the lower the loss, the closer your model is to ground truth so that is your target for optimization.

Here, though, there is no ground truth, there is only a reward function that you are trying to optimize for. The loss is just a direction for your optimization in which to optimize your parameters, as you are comparing the target Q to the network Q value, both of which are to be learned.

It does not matter if it is DQN or Policy gradient. What you learn in the end is to estimate the value of your state-action pair. The question for the network here is "if I take this action, what will be my current + future reward?". The more correct your Q value estimation is, the better the policy. Both methods solve this problem.

You could look into the implementation of a similar method with D3QN in: https://arxiv.org/pdf/1706.09829.pdf
You could take inspiration from their open source code.

closed 2024-01-06 by pzhabc