reiniscimurs/DRL-robot-navigation #78: The problem of resetting the vehicle and the end point at the end of a round.

hzxBuaa, 2023-10-30:
Hello, reiniscimurs. Could you help me answer the following questions? If you would answer this question, I would really appreciate it.

The problem of resetting the vehicle and the end point at the end of a round. How to ensure that the vehicle and the end point will not be reset into obstacles after a round ends? I see there is a check_pos function in it, but how do you get the poisition data?

reiniscimurs, 2023-10-30:
Hi,

For static obstacles we use `check_pos` function. This function essentially is a list of "dead zones" in the environment where starting and goal xy positions should not be generated. These are manually described based on the environment. We randomly sample robot and goal xy positions until they do not fall in any of the "dead zones". The position data is just a randomly sampled value. Only after we have ensured that it is a usable value, we reset the robot and goal position to this location.

You can read a more in-depth explanation here in Reset section: https://medium.com/@reinis_86651/deep-reinforcement-learning-in-mobile-robot-navigation-tutorial-part4-environment-7e4bc672f590

hzxBuaa, 2023-10-31:
Thank you very much！

closed 2023-11-08 by hzxBuaa