reiniscimurs/DRL-robot-navigation #79: Selection of local target points

ZCPRL, 2023-10-30:
Hello, I have not found the code of the part that selects the local target point in the project, may I ask where the code of this part is implemented?

reiniscimurs, 2023-10-30:
Goal point is set by calling the `change_goal` function: https://github.com/reiniscimurs/DRL-robot-navigation/blob/main/TD3/velodyne_env.py#L320

ZCPRL, 2023-10-30:
Thank you for such a quick reply. Your team's processing of laser data is to select points with the smallest distance from each Angle gap within 180 degrees and take these points as state values. My question is whether each Angle gap point is a local target point. Another problem is that the code does not see the calculation of the evaluation (h(ci)) of each local target point. According to the content of the paper, the point with the smallest h(ci) is selected as the local target point, looking forward to your answer.

[image]

reiniscimurs, 2023-10-30:
Hi,

> My question is whether each Angle gap point is a local target point.
I do not understand this question. What do you mean by local target point? Each gaps minimal laser reading is taken as state information. It has no bearing on goal position selection as it is a random point in the environment.

I think you might be looking at the wrong repository. This repo is just for training a generic DRL-based motion policy. The exploration algorithm (where a version of this policy is uesd) is available in the [GDAE](https://github.com/reiniscimurs/GDAE) repository. The calculation of POI is given in: https://github.com/reiniscimurs/IDLE-heuristic

pzhabc, 2023-11-02:
Hello, during the evaluation, when the robot is approaching the target point, it will stop there and keep rotating left and right, but it does not reach the target point, but when it is not in the evaluation period, it seems that there is no such problem. What is the situation?

closed 2024-03-08 by reiniscimurs