reiniscimurs/DRL-robot-navigation #82: 更改车辆模型  Change Vehicle Model

peipei518, 2023-11-09:
Thank you for your project. I would like to replace the robot model with my own four-wheel differential model, but I have found that when my robot collides after the replacement, the environment will not reset. I would like to ask the following question: 1. Is the collision detected in your code using radar point cloud data? I found that after replacing the model, I cannot find the topic of point cloud data in rviz, Could it be because of this that collisions cannot be detected. 2. Could it be some other problem that caused this result? 3.//velodyne_points Where did the points topic come from? After I changed the vehicle model, rostopic list was able to find this topic, but I cannot find it in rviz, and I am not using the radar xacro file from your project. Why is this?If possible, could you give me some suggestions.What work does it take to modify the vehicle model based on your project
[image]The picture shows the situation of my robot after collision

peipei518, 2023-11-09:
[image]
Sorry, the previous screenshot was uploaded incorrectly. This is my screenshot. After the collision, the rviz small and medium-sized car continued to move, and the radar data also shifted

reiniscimurs, 2023-11-09:
Hi,

What kind of changes did you make to the code itself? And what version of the repo are you using? 
If you changed the model and implementation, I would not be able to tell you where the sensor information comes from.

Yes, by default we use the velodyne lidar to detect collisions. If you have not connected this lidar to your robot, it will not be able to detect collisions.
https://github.com/reiniscimurs/DRL-robot-navigation/blob/main/TD3/velodyne_env.py#L182

What you can do is use your laser data as input to the collision detection function and update collision distance.

peipei518, 2023-11-10:
Thank you for your reply. I am using ros Noetic and tried to replace the radar with the original one, but there was still a collision without resetting the environment.I am still unable to choose the /velodyne_points topic in rviz, This may be due to my lack of understanding of the code. Here are the modifications I have made. If possible, could you please help me take a look at the problem.
import math
import os
import random
import subprocess
import time
from os import path

import numpy as np
import rospy
import sensor_msgs.point_cloud2 as pc2
from gazebo_msgs.msg import ModelState
from geometry_msgs.msg import Twist
from nav_msgs.msg import Odometry
from sensor_msgs.msg import PointCloud2
from squaternion import Quaternion
from std_srvs.srv import Empty
from visualization_msgs.msg import Marker
from visualization_msgs.msg import MarkerArray

# GOAL_REACHED_DIST = 0.4
GOAL_REACHED_DIST = 0.3
# COLLISION_DIST = 0.5
COLLISION_DIST = 0.35
TIME_DELTA = 0.1

# Check if the random goal position is located on an obstacle and do not accept it if it is
def check_pos(x, y):
    goal_ok = True

    if -3.8 > x > -6.2 and 6.2 > y > 3.8:
        goal_ok = False

    if -1.3 > x > -2.7 and 4.7 > y > -0.2:
        goal_ok = False

    if -0.3 > x > -4.2 and 2.7 > y > 1.3:
        goal_ok = False

    if -0.8 > x > -4.2 and -2.3 > y > -4.2:
        goal_ok = False

    if -1.3 > x > -3.7 and -0.8 > y > -2.7:
        goal_ok = False

    if 4.2 > x > 0.8 and -1.8 > y > -3.2:
        goal_ok = False

    if 4 > x > 2.5 and 0.7 > y > -3.2:
        goal_ok = False

    if 6.2 > x > 3.8 and -3.3 > y > -4.2:
        goal_ok = False

    if 4.2 > x > 1.3 and 3.7 > y > 1.5:
        goal_ok = False

    if -3.0 > x > -7.2 and 0.5 > y > -1.5:
        goal_ok = False

    if x > 4.5 or x < -4.5 or y > 4.5 or y < -4.5:
        goal_ok = False

    return goal_ok

class GazeboEnv:
    """Superclass for all Gazebo environments."""

    def __init__(self, launchfile, environment_dim):
        self.environment_dim = environment_dim
[... 351 lines cut ...]
        marker3.pose.position.x = 5
        marker3.pose.position.y = 0.2
        marker3.pose.position.z = 0

        markerArray3.markers.append(marker3)
        self.publisher3.publish(markerArray3)

    @staticmethod
    def observe_collision(laser_data):
        # Detect a collision from laser data
        min_laser = min(laser_data)
        if min_laser < COLLISION_DIST:
            return True, True, min_laser
        return False, False, min_laser

    @staticmethod
    def get_reward(target, collision, action, min_laser):
        if target:
            return 100.0
        elif collision:
            return -100.0
        else:
            r3 = lambda x: 1 - x if x < 1 else 0.0
            return action[0] / 2 - abs(action[1]) / 2 - r3(min_laser) / 2
In this code file, I only modified the vehicle name and cmd_ The topics of vel and odom have not been modified

peipei518, 2023-11-10:
<launch>
<arg name="gui" value="false"/>

   
      <include file="$(find gazebo_ros)/launch/empty_world.launch" >
       <arg name="world_name" value="$(find multi_robot_scenario)/launch/TD3.world" /> 
       
     </include>
  

  
  <include file="$(find autolabor_pro1_gazebo_worlds)/launch/autolabor_pro1_rplidar_gazebo.launch">
     
     
   </include>

  
  <node pkg="rviz" type="rviz" name="rviz" args="-d $(find multi_robot_scenario)/launch/autolabor.rviz" />

</launch>
This is the launch file where I started the robot model

peipei518, 2023-11-10:
This is the launch file where I started the robot model
[image]

peipei518, 2023-11-10:
<launch>
<arg name="gui" value="false"/>
      
           #####<include file="$(find gazebo_ros)/launch/empty_world.launch" >
         <arg name="world_name" value="$(find multi_robot_scenario)/launch/TD3.world" /> 
         
      </include>
      
      
      <include file="$(find autolabor_pro1_gazebo_worlds)/launch/autolabor_pro1_rplidar_gazebo.launch" />
      
      <node pkg="rviz" type="rviz" name="rviz" args="-d $(find multi_robot_scenario)/launch/autolabor.rviz" />
</launch>

peipei518, 2023-11-10:
This is my robot launch file
<launch>
    
    
    <arg name="paused" default="false"/>
    <arg name="use_sim_time" default="true"/>
    <arg name="gui" default="true"/>
    <arg name="headless" default="false"/>
    <arg name="debug" default="false"/>

    
    

    
    <param name="robot_description" command="$(find xacro)/xacro --inorder '$(find autolabor_description)/urdf/xacro/gazebo/autolabor_pro1_rplidar_gazebo.xacro'" /> 
  
    
    <node name="robot_state_publisher" pkg="robot_state_publisher" type="robot_state_publisher">
      <param name="publish_frequency" value="20.0"/>
    </node>

    
    <node name="joint_state_publisher" pkg="joint_state_publisher" type="joint_state_publisher">
      <param name="use_gui" value="True" />
      <param name="rate" value="20.0"/>
    </node>
    
    <node name="urdf_spawner" pkg="gazebo_ros" type="spawn_model" respawn="false" output="screen"
          args="-urdf -model autolabor_pro1 -param robot_description -x -0.1 -y -0.1 -z  0.1"/> 
</launch>

peipei518, 2023-11-10:
This is my robot xacro file, where I replaced the radar with the radar in your project, but the above problem still occurs
<?xml version="1.0"?>
<robot name="arm" xmlns:xacro="http://www.ros.org/wiki/xacro">

    <xacro:include filename="$(find autolabor_description)/urdf/autolabor_pro1.xacro" />
    
      <xacro:include filename="$(find multi_robot_scenario)/xacro/laser/hokuyo.xacro"/>
  <xacro:laser name="front_laser" parent="base_link" xyz="0 0 0.31" rpy="0 0 0" meshes="package://gazebo_plugins/test/multi_robot_scenario/meshes/laser"/>
    <xacro:include filename="$(find autolabor_description)/urdf/sensors/camera_gazebo.xacro" />

    <xacro:property name="camera_offset_x" value="0.09" />
    <xacro:property name="camera_offset_y" value="0" />
    <xacro:property name="camera_offset_z" value="0.31" />
    

    
    

    

    <joint name="camera_joint" type="fixed">
        <origin xyz="${camera_offset_x} ${camera_offset_y} ${camera_offset_z}" rpy="0 0 0" />
        <parent link="base_link"/>
        <child link="camera_link"/>
    </joint>

    <xacro:usb_camera prefix="camera"/>

    <autolabor_pro1/>

</robot>

peipei518, 2023-11-10:
I don't know why a few lines of code were lost during upload, but the main code is there. If possible, could you help me check where the problem occurred

reiniscimurs, 2023-11-10:
Your collision detection is still based on velodyne lidar input:
`# read velodyne laser state
    done, collision, min_laser = self.observe_collision(self.velodyne_data)`

You have not attatched velodyne to your robot so collisions will not be detected. If you want to just use the hokuyo laser, then you will have to read the laser state from this hokuyo lidar instead.

On P3DX robot the velodyne laser is added here:
https://github.com/reiniscimurs/DRL-robot-navigation/blob/main/catkin_ws/src/multi_robot_scenario/xacro/p3dx/pioneer3dx.xacro#L26-L29

peipei518, 2023-11-13:
Thank you for your reply. I have resolved the issue of changing the vehicle model radar topic, but there are currently some other issues. Could you please give me some suggestions again
1. My radar data seems to shift while the car is moving. When I change the Fixed Frame to World, it seems that the radar does not shift, but the target point will move with the car, which I believe is abnormal. Here is my tf_ Tree and rqt_ Graph.The first two images show the selection of Odom and World for Fixed Frame
[image]
[image]
[image]
[image]
2.Due to my larger vehicle model, I attempted to change the COLLISION_ The value of DIST has been increased, but there has been a constant reset. I found in your other replies that it seems to be a def check_ The issue with pos (x, y), but I don't know the basis for setting the parameters in this function, so I don't know how to change it< Box size="0.7248 0.4302 0.195"/>These are the parameters of my chassis. Could you please give me some suggestions.

reiniscimurs, 2023-11-13:
1. This would most likely be something dependent on your implementation and setup between the robot and the world. If setting fixed frame to world works for you, you can try publishing the goal points and markers in that frame as well. By default they are published in odom frame : https://github.com/reiniscimurs/DRL-robot-navigation/blob/main/TD3/velodyne_env.py#L366

2. You collision is quite obviously detected from your own robot. Meaning that the laser detects your own robot base, and since the laser values are so close, reset is triggered. The function `check_pos` will not have any influence on this and is used only for placing a random goal position. It has nothing to do with triggering reset. I do not see the robot model clearly, but it seems that you have also placed velodyne lidar sensor on top of your 2D lidar. This could also cause collisions to be detected. In any case, the issue is caused due to suboptimal placement of your lidar sensors.

I would suggest going through the code and really understanding what each function does in detail. It will be really difficult making changes without understanding what parts of the code take care of what. There is a code guide provided and it should help you understand the meaning of functions and their use.

peipei518, 2023-11-14:
hi  I am using a branch of Melodic
With your suggestion, I set the fixed frame to world and made changes in the code, as well as changing the position of the 3D LiDAR. Now the collision can reset the environment normally. Due to my larger robot model, I changed the min in the code_ Range set to 0.6, Dist<0.6. Now my radar data and target point do not follow the deviation of the car, but I have found that the TF of the odom will shift. I cannot understand the reason for this. Now, my car passes through the target point but does not reset the environment. I suspect it is due to the deviation of the odom. My tf_ The tree is the same as above. If possible, could you please give me some suggestions? Thank you again!
[image]
Here is my modified velodyne_ Env.py code，
import rospy
import subprocess
from os import path
from visualization_msgs.msg import Marker
from visualization_msgs.msg import MarkerArray
from numpy import inf
import numpy as np
import random
import math
from gazebo_msgs.msg import ModelState
from squaternion import Quaternion
from geometry_msgs.msg import Twist
from sensor_msgs.msg import LaserScan, PointCloud2
import sensor_msgs.point_cloud2 as pc2
from nav_msgs.msg import Odometry
from std_srvs.srv import Empty
import os
import time

os.environ['TF_CPP_MIN_LOG_LEVEL'] = '3'

# Check if the random goal position is located on an obstacle and do not accept it if it is
def check_pos(x, y):
    goalOK = True

    if -3.8 > x > -6.2 and 6.2 > y > 3.8:
        goalOK = False

    if -1.3 > x > -2.7 and 4.7 > y > -0.2:
        goalOK = False

    if -0.3 > x > -4.2 and 2.7 > y > 1.3:
        goalOK = False

    if -0.8 > x > -4.2 and -2.3 > y > -4.2:
        goalOK = False

    if -1.3 > x > -3.7 and -0.8 > y > -2.7:
        goalOK = False

    if 4.2 > x > 0.8 and -1.8 > y > -3.2:
        goalOK = False

    if 4 > x > 2.5 and 0.7 > y > -3.2:
        goalOK = False

    if 6.2 > x > 3.8 and -3.3 > y > -4.2:
        goalOK = False

    if 4.2 > x > 1.3 and 3.7 > y > 1.5:
        goalOK = False

    if -3.0 > x > -7.2 and 0.5 > y > -1.5:
        goalOK = False

    if x > 4.5 or x < -4.5 or y > 4.5 or y < -4.5:
        goalOK = False
    return goalOK

# Function to put the laser data in bins
def binning(lower_bound, data, quantity):
    width = round(len(data) / quantity)
    quantity -= 1
    bins = []
    for low in range(lower_bound, lower_bound + quantity * width + 1, width):
        bins.append(min(data[low:low + width]))
[... 405 lines cut ...]
    def random_box(self):
        for i in range(4):
            name = 'cardboard_box_' + str(i)

            x = 0
            y = 0
            chk = False
            while not chk:
                x = np.random.uniform(-6, 6)
                y = np.random.uniform(-6, 6)
                chk = check_pos(x, y)
                d1 = math.sqrt((x - self.odomX) ** 2 + (y - self.odomY) ** 2)
                d2 = math.sqrt((x - self.goalX) ** 2 + (y - self.goalY) ** 2)
                if d1 < 1.5 or d2 < 1.5:
                    chk = False
            box_state = ModelState()
            box_state.model_name = name
            box_state.pose.position.x = x
            box_state.pose.position.y = y
            box_state.pose.position.z = 0.
            box_state.pose.orientation.x = 0.0
            box_state.pose.orientation.y = 0.0
            box_state.pose.orientation.z = 0.0
            box_state.pose.orientation.w = 1.0
            self.set_state.publish(box_state)

reiniscimurs, 2023-11-14:
Hi,

Melodic version of this repo is deprecated and is not supported from my side anymore. As such it is only provided "as is" and any changes you make should be up to you. However, the code itself should not impact your frame drift. Changing the collision distance would not change it either.

Most likely, you need to check your robot implementation and how you are setting up the transforms between different frames. But this is something you should be doing on your side as it is an unfamiliar issue and implementation for me. I do not see any obvious issues here, so I cannot help you with advice here either. Hope you manage to solve the issue.

peipei518, 2023-11-14:
Thank you for your reply. The problem I am currently facing is that the collision reset environment can be achieved, but when the car reaches the target point, the environment will not be reset. Is this caused by the deviation of my odom or other factors? Can you give me some suggestions? Thank you!

reiniscimurs, 2023-11-14:
Most likely the target point is actually not arrived at. Since you are displaying in a different frame than odom, it might appear in rviz that the goal is reached. But the reached goal trigger is the relevant distance between the goal and the robot in odom frame. You should print out the distance values to confirm that distance actually never goes below the threshold of triggering a reached goal.

I would guess that something is misaligned with your frames. Unfortunately, I cannot help you there as I have no experience in implementing the robot you are trying to use.

peipei518, 2023-11-15:
Thank you for your reply. With your suggestion, I attempted to print the distance value from the target point and found that, as you mentioned, when I selected World, it was displayed as arrived but not actually arrived. When I selected Odom, although the radar data would shift with the car's movement, it could still reach the target point normally and reset the environment. I tried debugging my code for a day, but I couldn't find the correct direction. I tried changing<publishWorldTF>true</publishWorldTF>to false, but the radar data was still offset when choosing odom. I also tried changing to false and added the tf transformation from world to odom from the launch file. As a result, the radar data also encountered the above problem when choosing world, What I am currently wondering is why radar data is only normal when choosing World. Is it defined in my xacro file?
I will provide you with the xacro file of my car model, and if possible, I would like to receive some suggestions from you.
This is my plugin file
<?xml version="1.0"?>
<robot>

     <gazebo reference="base_footprint">
        <mu1>0.3</mu1>
        <mu2>0.3</mu2>
    </gazebo>
    <gazebo reference="base_link">
        <mu1>0.3</mu1>
        <mu2>0.3</mu2>
    </gazebo>

    <gazebo reference="left_front_wheel">
    	<mu1>1.5</mu1>
    	<mu2>1.5</mu2>
    </gazebo>

    <gazebo reference="left_back_wheel">
    	<mu1>1.5</mu1>
    	<mu2>1.5</mu2>
    </gazebo>

    <gazebo reference="right_front_wheel">
    	<mu1>1.5</mu1>
    	<mu2>1.5</mu2>
    </gazebo>

    <gazebo reference="right_back_wheel">
    	<mu1>1.5</mu1>
    	<mu2>1.5</mu2>
    </gazebo>

    <gazebo>
        <plugin name="autolabor_driver_gazebo_plugin" filename="libfour_wheel_diff_driver_plugin.so">
            <rosDebugLevel>Debug</rosDebugLevel>
            <commandTopic>cmd_vel</commandTopic>
            <odometryTopic>odom</odometryTopic>
            
            <odometryFrame>odom</odometryFrame>
            <odometrySource>world</odometrySource>
            <robotBaseFrame>base_link</robotBaseFrame>
            <worldFrame>world</worldFrame>

            <publishWheelTF>false</publishWheelTF>
            <publishOdomTF>true</publishOdomTF>
            <publishWorldTF>true</publishWorldTF>
            <publishWheelJointState>false</publishWheelJointState>
            <legacyMode>false</legacyMode>

            <wheelSeparation>0.80</wheelSeparation>
            <wheelDiameter>0.25</wheelDiameter>
            <wheelAcceleration>0.0</wheelAcceleration>
            <wheelTorque>100.0</wheelTorque>
            <updateRate>10.0</updateRate>

            <leftFrontWheelJoint>joint_left_front</leftFrontWheelJoint>
            <leftBackWheelJoint>joint_left_back</leftBackWheelJoint>
            <rightFrontWheelJoint>joint_right_front</rightFrontWheelJoint>
            <rightBackWheelJoint>joint_right_back</rightBackWheelJoint>
        </plugin>
    </gazebo>

    <gazebo reference="imu_link">
        <gravity>true</gravity>
        <sensor name="imu_sensor" type="imu">
            <always_on>true</always_on>
            <update_rate>20</update_rate>
            <visualize>true</visualize>
            <topic>imu</topic>
            <plugin filename="libimu_plugin.so" name="imu_plugin">
                <topicName>imu</topicName>
                <bodyName>imu_link</bodyName>
                <updateRateHZ>10</updateRateHZ>
                <gaussianNoise>0.0</gaussianNoise>
                <xyzOffset>0 0 0</xyzOffset>
                <rpyOffset>0 0 0</rpyOffset>
                <frameName>imu_link</frameName>
                <enableOrientation>true</enableOrientation>
                <initialOrientationAsReference>false</initialOrientationAsReference>
            </plugin>
        </sensor>
    </gazebo>

    <gazebo reference="lider_1_link">
        <sensor name="sensor_ray" type="gpu_ray">
        
            <ray>
                <scan display="true">
                    <horizontal>
                        <samples>500</samples>
                        <resolution>1.0</resolution>
                        <min_angle>-1.0</min_angle>
                        <max_angle>1.0</max_angle>
                    </horizontal>
                    <vertical>
                        <samples>160</samples>
                        <resolution>1.0</resolution>
                        <min_angle>-0.1</min_angle>
                        <max_angle>0.1</max_angle>
                    </vertical>
                </scan>
                <range>
                    <min>0.05</min>
                    <max>160.0</max>
                </range>
            </ray>

            <plugin name="plugin_1" filename="libblocklaser_plugin.so">
                <gaussianNoise>0.00</gaussianNoise>
                <alwaysOn>true</alwaysOn>
                <updateRate>10</updateRate>
                <topicName>points2</topicName>
                <frameName>lider_1_link</frameName>
            </plugin>
            <always_on>true</always_on>
            <update_rate>10</update_rate>
        </sensor>
    </gazebo>
</robot>
This is my vehicle's xacro file
[image]
[image]
[image]
[image]

peipei518, 2023-11-15:
[image]
[image]
[image]
[image]

reiniscimurs, 2023-11-15:
Sorry, this is outside the scope of this repo. I do not know what your full robot setup is like, I cannot help you with that. All I can say is that I do not understand what is this `world` frame is and why `base_link`  has a transform aiming towards it. This seems strange. Most likely, your transforms are set up wrong. But as I said, this is not something I can help you with.

peipei518, 2023-11-17:
Thank you very much for your help. I found that the problem was caused by the Gazebo driver plugin. I replaced the plugin and the problem was resolved

closed 2023-11-17 by peipei518