reiniscimurs/DRL-robot-navigation #84: test the training results

peipei518, 2023-11-10:
Hello, I downloaded the Melodic branch of your project, but I couldn't find the code file for testing the training results in it. If I use the Noetic version of the testing code, I will report an error. How can I test the training results

peipei518, 2023-11-10:
I use the Noetic version of the testing code，The terminal will output the following error

/home/autolabor/.local/lib/python3.6/site-packages/torch/cuda/__init__.py:80: UserWarning: CUDA initialization: The NVIDIA driver on your system is too old (found version 9010). Please update your GPU driver by downloading and installing a new version from the URL: http://www.nvidia.com/Download/index.aspx Alternatively, go to: https://pytorch.org to install a PyTorch version that has been compiled with your version of the CUDA driver. (Triggered internally at  ../c10/cuda/CUDAFunctions.cpp:112.)
  return torch._C._cuda_getDeviceCount() > 0
Traceback (most recent call last):
  File "test_velodyne_td3.py", line 55, in <module>
    env = GazeboEnv("multi_robot_scenario.launch", environment_dim)
TypeError: __init__() missing 2 required positional arguments: 'width' and 'nchannels'

reiniscimurs, 2023-11-10:
There is a discrepancy between how the `GazeboEnv` is initialized in noetic and melodic. This is purely a simple python issue, you are missing input values for the class instantiation. AFAIK you could put anything there as they are not used in the code anyway, which is why it was removed in the noetic version.

peipei518, 2023-11-13:
Thank you very much. With your suggestion, I have solved the problem very well

closed 2023-11-13 by peipei518