reiniscimurs/DRL-robot-navigation #90: There is a problem replacing the robot model

pzhabc, 2023-12-10:
Hello, thank you for sharing. After I changed the model of the robot to "turtlebot3_burger", the robot has been in the same place, can you tell me the reason? Thank you very much!

pzhabc, 2023-12-10:
After a while, this happened.
https://github.com/reiniscimurs/DRL-robot-navigation/assets/147294035/d0c529fd-f758-48f6-835e-056bf9fa1c7c
[image]

pzhabc, 2023-12-10:
I've found a few reasons why using this code is fine.
[image]
Using this code will cause problems.
[image]

reiniscimurs, 2023-12-10:
Hi,

Please provide full information of the changes you have made and what exactly you are trying to do. It is very difficult to give any reasonable suggestions without more information.

pzhabc, 2023-12-11:
> 你好
> 
> 请提供您所做的更改以及您到底要做什么的完整信息。如果没有更多信息，很难给出任何合理的建议。

> Hi,
> 
> Please provide full information of the changes you have made and what exactly you are trying to do. It is very difficult to give any reasonable suggestions without more information.

I used someone else's velodyne_env.py, who is quoting your code, and when I changed the robot model from "Pionner3dx" to "turtlebot3_burger", the robot stopped moving. My initial judgment was that it was a lidar problem. The problem arises when I replace the liDAR with the following:
[image]
With this, there's no problem：
[image]
Here is the main section of "velodyne_env.py" :
        self.unpause = rospy.ServiceProxy('/gazebo/unpause_physics', Empty)
        self.pause = rospy.ServiceProxy('/gazebo/pause_physics', Empty)
        self.reset_proxy = rospy.ServiceProxy('/gazebo/reset_world', Empty)
    
        self.publisher = rospy.Publisher('vis_mark_array', MarkerArray, queue_size=10)  
        self.vel_pub = rospy.Publisher('/r1/cmd_vel', Twist, queue_size=1)
        self.set_state = rospy.Publisher('gazebo/set_model_state', ModelState, queue_size=10)   

        self.velodyne = rospy.Subscriber('/velodyne_points', PointCloud2, self.velodyne_callback, queue_size=10)   
        self.laser = rospy.Subscriber('/r1/front_laser/scan', LaserScan, self.laser_callback, queue_size=10)    
        self.odom = rospy.Subscriber('/r1/odom', Odometry, self.odom_callback, queue_size=10)

pzhabc, 2023-12-11:
> Which repository branch are you using? You mention changing the p3dx model, why not just use the turtlebot branch and then just call the burger model?
> 
> Where is this code that you added from? It seems to me it is describing only joints of the sensor and not the sensor itself. Do you get any topics published as a laser scan? If you are using melodic branch, the robot will wait for scan to become available. If there is no scan, it will do nothing.
> 
> Just to re-iterate: Please provide full information about your problem and your setup such as (but not limited to) repo branch, your system setup, where did you get the other repo from, what exact code changes you made. Even if you think it is self explanatory, I cannot know what you are working with so please provide any sort of information you can give. Or else it is really difficult to help with any issues.

I am using this code:
[env.txt](https://github.com/reiniscimurs/DRL-robot-navigation/files/13636446/env.txt)
[image]
[image]

[turtlebot3_burger.urdf.txt](https://github.com/reiniscimurs/DRL-robot-navigation/files/13636635/turtlebot3_burger.urdf.txt)
[turtlebot3_burger.gazebo.txt](https://github.com/reiniscimurs/DRL-robot-navigation/files/13636648/turtlebot3_burger.gazebo.txt)

reiniscimurs, 2023-12-14:
Which version of the repo are you using and do you see your laser topic being published?

pzhabc, 2023-12-14:
I saw two lidar topics in the terminal, one was in "turtlebot3_burger.gazebo.xacro" and one was in "env.py".What is the reason for this?
[image]
[image]

reiniscimurs, 2023-12-14:
Please provide the information asked.

I cannot help you if you do not provide meaningful information. If you do not know what information to provide, then please read previous comments or ask. You have changed some parts of the code that I do not know and are asking questions about it. I cannot help you in such a case.

pzhabc, 2023-12-14:
> Please provide the information asked.
>
> I cannot help you if you do not provide meaningful information. If you do not know what information to provide, then please read previous comments or ask. You [...]

Okay, the main thing is that the 'env.txt' file I provided above has been modified, and the other files are from your "turtlebot3" library. I can't seem to find this at the moment. I just copied the Pioneer3dx robot model from this library. , changed to the turtlebot3 burger robot model, and other files were not modified. But when I directly replaced it with turtlebot3 burger, an error occurred. I wonder if this information is enough? Sorry for the inconvenience.

reiniscimurs, 2023-12-14:
I am still unsure what your repository is that you are using. Is it the Noetic-Turtlebot branch? But there are changes added from Melodic branch?

And what you are trying to do is to add 2d laser to the robot?
It seems to me that you are doing something very messy here and I find it very difficult to understand what is what.

In any case your laser scan topic names do not line up. Check what is the name of the laser sensor that you are actually publishing.

closed 2024-03-08 by reiniscimurs