reiniscimurs/DRL-robot-navigation #92: Some questions about sim2real

nlnlnl1, 2023-12-21:
First of all, thank you very much for sharing.
May I ask if it is necessary to perform some coordinate system transformation on the input of the Actor network when deploying the trained model on a real car？
I tried to build the same Actor network structure on a physical car and then load the weight file trained in a virtual environment. The four inputs for robots are their linear velocity, angular velocity, heading angle, and distance. Because I observed in rviz that the coordinates of the car were read in a Cartesian coordinate system. But the position of the real car is read in the right-hand coordinate system. So when calculating the heading angle, do I need to convert the coordinate system, or how should I convert it on a real car?
I'm not sure if I made it clear, but in the real car section, my real car can move normally but cannot reach the target point.So I really need your advice and help, and I look forward to your reply.

reiniscimurs, 2023-12-21:
The actor network itself is agnostic to what the real coordinate frame is in the environment. What it requires is an input in polar coordinates. Meaning that the goal position needs to be given in polar coordinates respective to the current heading direction of the robot. See more in depth explanation in: https://medium.com/@reinis_86651/deep-reinforcement-learning-in-mobile-robot-navigation-tutorial-part3-training-13b2875c7b51

I do not know your real world setup so I would not know what to suggest. But the first thing would be to make sure that the angle and distance information makes sense as an input to the actor network.

nlnlnl1, 2023-12-25:
Thank you for your reply. I understand what you mean now. No matter what coordinate system is used, we only need to calculate the relative heading angle of the position information in the polar coordinate system and input it into the actor model.

closed 2024-03-08 by reiniscimurs