reiniscimurs/DRL-robot-navigation #94: Problem with the starting training

Ethan0207, 2023-12-25:
First of all, thank you for your sharing. This a great project .I am trying to train the agent you created with the codes you provided and I did everything as you mentioned. But I have a problem about it. When I start training, it just start only “roscore launched”.I have no idea. Is there something wrong with the environment?
[image]

reiniscimurs, 2023-12-25:
Hi,

Yes, seems there is an issue with your ros launch. The program should automatically launch roscore and you should see the logs.
Which repo version and which ros version are you using?

Ethan0207, 2023-12-25:
Thank you very much for your reply. My ROS version is 20.04. But repo has not been installed.  Does this need to be installed?

reiniscimurs, 2023-12-25:
Repo does not need to be installed, just compiled. It does however have 3 different branches and I would need to know which branch you have cloned.

What seems to be the issue that roscore is not being launched. Can you launch roscore on its own in an empty terminal?

Ethan0207, 2023-12-25:
Thank you for your reply. Firstly, I have cloned "main". And I can launch roscore on its own in an empty terminal
[image]
[image]

reiniscimurs, 2023-12-25:
Seems to me that `subprocess.Popen(["roscore", "-p", port])` is not working in your setup. You could test it in a dummy script or debug in the current setup.

Alternatively, you can manually start roscore and the gazebo before running the training script.

Ethan0207, 2023-12-28:
Hello，thank you for your reply. These days, I try to solve my problems , but it still has some mistakes .These mistakes are similar to the issue of "Problem with the starting training" mentioned by the friend below.
When I try to do it without exporting this commands;
export ROS_HOSTNAME=localhost
export ROS_MASTER_URI=http://localhost:11311/
export ROS_PORT_SIM=11311
export GAZEBO_RESOURCE_PATH=~<PATH_TO_FOLDER>/DRL-robot-navigation/catkin_ws/src/multi_robot_scenario/launch
source ~/.bashrc
cd ~<PATH_TO_FOLDER>/DRL-robot-navigation/catkin_ws
source devel_isolated/setup.bash
In my terminal . it says roscore on the top and nothing happens.
[image]
if I export this
export ROS_HOSTNAME=localhost
export ROS_MASTER_URI=http://localhost:11311/
export ROS_PORT_SIM=11311
export GAZEBO_RESOURCE_PATH=~<PATH_TO_FOLDER>/DRL-robot-navigation/catkin_ws/src/multi_robot_scenario/launch
source ~/.bashrc
cd ~<PATH_TO_FOLDER>/DRL-robot-navigation/catkin_ws
source devel_isolated/setup.bash
In my terminal . it still says roscore on the top and nothing happens.
[image]
So I am stuck at this point and I can't train the agent . If you can help me with this problem I would be so glad. Thanks in advance.
additionally this is my .bashrc file
[image]

closed 2024-04-28 by reiniscimurs