reiniscimurs/DRL-robot-navigation #95: Some problems

pzhabc, 2023-12-26:
Hello.These codes were removed by me:
[image]
But there's still this in the topic:
[image]
I am running directly "multi_robot_scenario.launch". What is the reason for this?

reiniscimurs, 2023-12-26:
Have you also removed the subscriber to this topic? 
https://github.com/reiniscimurs/DRL-robot-navigation/blob/main/TD3/velodyne_env.py#L126

Most likely, this is where this topic is coming from.

pzhabc, 2023-12-27:
> Have you also removed the subscriber to this topic? https://github.com/reiniscimurs/DRL-robot-navigation/blob/main/TD3/velodyne_env.py#L126
> 
> Most likely, this is where this topic is coming from.

Hello, one more question, I haven't seen everything related to the /scan topic, but why does this topic exist? It feels like a strange situation. I've tried restarting vscode and the computer, but it doesn't seem to help.
[image]
[image]
[image]
[image]
[image]

reiniscimurs, 2023-12-27:
There is no `/scan` topic in the original code. That must be something that you have added with your changes.

You should use the search functions in your IDE to find the source of your topic.

closed 2024-01-06 by pzhabc