reiniscimurs/DRL-robot-navigation #96: Problems with modifying the model

cd310105974, 2023-12-28:
Hi Reinis Cimurs, first of all, thank you very much for sharing.
I tried modifying your model into an Ackermann car. There is no error when launching but the car still does not move. The running picture is as follows. I hope you can give me some suggestions to solve the problem. 
Looking forward to your reply.
[image]
[image]

reiniscimurs, 2023-12-28:
Hi,

I see you are using ROS melodic distribution in your code. This is a version I am no longer supporting as I have no way of testing it so generally any issues arising in this distribution would be up to you to solve.

That being said, I can take a brief look if you share the code changes that you have implemented. I don't really see anything just from the log messages.

cd310105974, 2023-12-28:
Thank you for your reply.
In addition to changing your model to Ackerman, I also made a few modifications:
1. Using the 'libgazebo_ros_control.so' plugin and 'transmission' to control the car.
[image]

2. Setting up the controller.
[image]

3. Calculating the speed and assign it to the controller command.
[image]
[image]

4. Posting the odom topics and the tf.
[image]
[image]

This is my launch file.
[image]

These are the modifications I made to the code.

reiniscimurs, 2023-12-28:
I just took a quick look but I suggest checking ros topics and the names for them. It seems to me you are using a namespace in the robot controller plugin that was not used in the default (or it was passed in instead of hardcoding it). So if your cmd_vel topic is in a different namesake, the publisher will not be able to send the message to the controller and the robot will not move. See the outputs from "rostopic list". Also would be good to see RQT tree and published nodes.

cd310105974, 2023-12-29:
Thank you very much for your reply.
After inspection, the car was finally ready to drive. The problem is the namespace. It is not easy for a newbie. 
However, I discovered a problem. In rviz, the laser behind the car will move. This problem also occurs with the gif you uploaded. Have you analyzed the reason?

reiniscimurs, 2023-12-29:
What do you mean by "laser behind the car"? We do not have a laser in the main repo branch at all, just the velodyne lidar,and there is no laser behind the robot in other branches. If you mean the swaying of the laser data in the gif, that is due to running the training with increased simulation speed as described in: https://medium.com/@reinis_86651/deep-reinforcement-learning-in-mobile-robot-navigation-tutorial-part5-some-extra-stuff-b744852345ac
There are some drawbacks to increasing the speed.

closed 2024-04-28 by reiniscimurs