reiniscimurs/DRL-robot-navigation #97: Problem with  training

Ethan0207, 2023-12-28:
Hello. These days, I try to solve my problems , but it still has some mistakes .These mistakes are similar to the issue of "Problem with the starting training" mentioned by the friend below.
When I try to do it without exporting this commands;
export ROS_HOSTNAME=localhost
export ROS_MASTER_URI=http://localhost:11311/
export ROS_PORT_SIM=11311
export GAZEBO_RESOURCE_PATH=~<PATH_TO_FOLDER>/DRL-robot-navigation/catkin_ws/src/multi_robot_scenario/launch
source ~/.bashrc
cd ~<PATH_TO_FOLDER>/DRL-robot-navigation/catkin_ws
source devel_isolated/setup.bash
In my terminal . it says roscore on the top and nothing happens.
[image]
if I export this
In my terminal . it still says roscore on the top and nothing happens.
[image]
So I am stuck at this point and I can't train the agent . If you can help me with this problem I would be so glad. Thanks in advance.
additionally this is my .bashrc file
[image]

reiniscimurs, 2023-12-28:
Hi,

Is this the first time starting the training? You might have missing gazebo models that the simulator is trying to load. What could be happening is that gazebo tries to load the world file and the models in it, but you do not have all the models available locally, so gazebo will download them. This takes some time and there are no indicators for this. 

You could try to open the world file directly in gazebo.
Alternatively, just start training and wait for some 20 to 30 minutes (it will download the models in the background), then see if the execution has started.

If this does not help, let me know.

Ethan0207, 2023-12-29:
Hi，thank you for your reply. Today, I just start training and wait for some 20 to 30 minutes, but it has not start. And then I  try to open the world file directly in gazebo.
[image]
[image]
[image]
(additionally this is my TD_world.launch file)
[image]
but it still has not start.
[image]
And I find annother question. When I "roslaunch multi_robot_scenario pioneer3dx.gazebo.launch", it just stop here.
[image]
I must "conda deactivate" ,then the p3dx appears. Do you think that's a contributing factor?

cd310105974, 2023-12-29:
The previously running node was not completely terminated. To kill the training process:
`killall -9 rosout roslaunch rosmaster gzserver nodelet robot_state_publisher gzclient python python3`

reiniscimurs, 2023-12-29:
Hi,

as @cd310105974 pointed out the error message in one of the images most likely is because you did not properly kill your previous run.

For the main issue though, i would suggest to see if the way to source the locations is proper in your conda env as well. Especially, if you need to use "localhost". I can see that your gazebo simulator is not starting from calling it in the training script and the line `subprocess.Popen(["roslaunch", "-p", port, fullpath])` does not execute for you.

Ethan0207, 2024-01-02:
Hi,
thank you for your reply. Firstly, I kill my previous run, but it is still failed.  I delete the commands in my bashrc file about conda and redo the steps . My ros still doesn't start. It is strange.
[image]

reiniscimurs, 2024-01-02:
I'd suggest trying to set up the training without using a virtual env to see if it is specifically virtual env issue. Seems that there is some issue with setting up and running the repo in virtual env. I won't be able to set that up and test it anytime soon though.

Ethan0207, 2024-01-03:
Thank you for your reply .I have solved the problem. I uninstall the Anaconda and install pytorch and tensorboard. It can work. I guess the version of ros and Anaconda clashed.
[image]

closed 2024-04-28 by reiniscimurs