reiniscimurs/DRL-robot-navigation #98: Gazebo environment

zichengS1999, 2024-01-03:
Is the simulation environment you created yourself or an existing open-source simulation environment?

zichengS1999, 2024-01-06:
Thank you for your reply! Is this world file open-source and available for use? Can I use it to test my research direction, and can I cite it in my paper? Additionally, I have encountered an issue during my experiments where the robot in Rviz becomes unsynchronized and randomly teleports when I call the reset() function. Have you encountered a similar situation before?

https://github.com/reiniscimurs/DRL-robot-navigation/assets/116474299/6aada16d-217b-42c5-a8c3-a44b7b193595

zichengS1999, 2024-01-06:
Thank you for your reply! Is this world file open-source and available for use? Can I use it to test my research direction, and can I cite it in my paper? Additionally, I have encountered an issue during my experiments where the robot in Rviz becomes unsynchronized and randomly teleports when I call the reset() function. Have you encountered a similar situation before?

reiniscimurs, 2024-01-06:
You need to check the licenses in this and their repo, but you can use this for research. And yes, you should cite the paper where the world file is taken from.

Not sure what you mean by unsynchronized. Each reset call will randomly teleport the robot to a new location and star executing a new episode. That is how training is intended and done on purpose.

zichengS1999, 2024-01-08:
The issue I encountered is the same as the first issue. It is like the two images below, where the robot's position changes from the first image to the second image after calling reset(). I also changed the robot and used a 2D laser.

[image]
[image]

reiniscimurs, 2024-01-08:
Most likely it is an issue with transforms and might just be an issue with visualization in rviz. What is your fixed frame in rviz (under global options)? By default we use odom, but it might be different in your case if you have changed the robot model. You would have to look into your code to see what the proper transforms are.

closed 2024-04-28 by reiniscimurs

wxy666ustb, 2025-05-08:
Hello, I have also encountered this problem. May I ask how you solved it?