reiniscimurs/DRL-robot-navigation #99: catkin_make_isolated error

sympetrumz, 2024-01-06:
It shows a lot of error before it catkin_make failed.

[image]

I use Ubuntu 20.04 
ROS Noetic
Python 3.8

sympetrumz, 2024-01-06:
/usr/include/gazebo-11/gazebo/msgs/tactile.pb.h:130:38: error: expected primary-expression before ‘>’ token
  130 |     return CreateMaybeMessage<Tactile>(arena);
      |                                      ^
/usr/include/gazebo-11/gazebo/msgs/spherical_coordinates.pb.h: In member function ‘virtual gazebo::msgs::SphericalCoordinates* gazebo::msgs::SphericalCoordinates::New(google::protobuf::Arena*) const’:
/usr/include/gazebo-11/gazebo/msgs/spherical_coordinates.pb.h:148:12: error: ‘CreateMaybeMessage’ was not declared in this scope
  148 |     return CreateMaybeMessage<SphericalCoordinates>(arena);
      |            ^~~~~~~~~~~~~~~~~~
/usr/include/gazebo-11/gazebo/msgs/spherical_coordinates.pb.h:148:51: error: expected primary-expression before ‘>’ token
  148 |     return CreateMaybeMessage<SphericalCoordinates>(arena);
      |                                                   ^
/usr/include/gazebo-11/gazebo/msgs/tactile.pb.h: In member function ‘virtual int gazebo::msgs::Tactile::GetCachedSize() const’:
/usr/include/gazebo-11/gazebo/msgs/tactile.pb.h:146:44: error: ‘_cached_size_’ was not declared in this scope
  146 |   int GetCachedSize() const final { return _cached_size_.Get(); }
      |                                            ^~~~~~~~~~~~~
/usr/include/gazebo-11/gazebo/msgs/spherical_coordinates.pb.h: In member function ‘virtual int gazebo::msgs::SphericalCoordinates::GetCachedSize() const’:
/usr/include/gazebo-11/gazebo/msgs/spherical_coordinates.pb.h:164:44: error: ‘_cached_size_’ was not declared in this scope
  164 |   int GetCachedSize() const final { return _cached_size_.Get(); }
      |                                            ^~~~~~~~~~~~~
In file included from /usr/include/gazebo-11/gazebo/msgs/MessageTypes.hh:114,
                 from /usr/include/gazebo-11/gazebo/msgs/msgs.hh:35,
                 from /usr/include/gazebo-11/gazebo/physics/Entity.hh:28,
                 from /usr/include/gazebo-11/gazebo/physics/Model.hh:30,
                 from /usr/include/gazebo-11/gazebo/physics/Actor.hh:27,
                 from /usr/include/gazebo-11/gazebo/physics/physics.hh:2,
                 from /home/symzpc/DRL-robot-navigation/catkin_ws/src/velodyne_simulator/velodyne_gazebo_plugins/include/velodyne_gazebo_plugins/GazeboRosVelodyneLaser.h:49,
                 from /home/symzpc/DRL-robot-navigation/catkin_ws/src/velodyne_simulator/velodyne_gazebo_plugins/src/GazeboRosVelodyneLaser.cpp:35:
/usr/include/gazebo-11/gazebo/msgs/subscribe.pb.h: At global scope:
/usr/include/gazebo-11/gazebo/msgs/subscribe.pb.h:244:41: error: ‘CachedSize’ in namespace ‘google::protobuf::internal’ does not name a type; did you mean ‘ToCachedSize’?
  244 |   mutable ::google::protobuf::internal::CachedSize _cached_size_;
      |                                         ^~~~~~~~~~
      |                                         ToCachedSize
/usr/include/gazebo-11/gazebo/msgs/tactile.pb.h: In member function ‘gazebo::msgs::Time* gazebo::msgs::Tactile::mutable_time()’:
/usr/include/gazebo-11/gazebo/msgs/tactile.pb.h:409:15: error: ‘CreateMaybeMessage’ was not declared in this scope
  409 |     auto* p = CreateMaybeMessage<::gazebo::msgs::Time>(GetArenaNoVirtual());
      |               ^~~~~~~~~~~~~~~~~~
/usr/include/gazebo-11/gazebo/msgs/tactile.pb.h:409:54: error: expected primary-expression before ‘>’ token
  409 |     auto* p = CreateMaybeMessage<::gazebo::msgs::Time>(GetArenaNoVirtual());
      |                                                      ^
/usr/include/gazebo-11/gazebo/msgs/tactile.pb.h: In member function ‘void gazebo::msgs::Tactile::set_allocated_time(gazebo::msgs::Time*)’:
/usr/include/gazebo-11/gazebo/msgs/tactile.pb.h:423:44: error: ‘GetOwnedMessage’ is not a member of ‘google::protobuf::internal’
  423 |       time = ::google::protobuf::internal::GetOwnedMessage(
      |                                            ^~~~~~~~~~~~~~~
In file included from /usr/include/gazebo-11/gazebo/msgs/MessageTypes.hh:117,
                 from /usr/include/gazebo-11/gazebo/msgs/msgs.hh:35,
                 from /usr/include/gazebo-11/gazebo/physics/Entity.hh:28,
                 from /usr/include/gazebo-11/gazebo/physics/Model.hh:30,
                 from /usr/include/gazebo-11/gazebo/physics/Actor.hh:27,
                 from /usr/include/gazebo-11/gazebo/physics/physics.hh:2,
                 from /home/symzpc/DRL-robot-navigation/catkin_ws/src/velodyne_simulator/velodyne_gazebo_plugins/include/velodyne_gazebo_plugins/GazeboRosVelodyneLaser.h:49,
                 from /home/symzpc/DRL-robot-navigation/catkin_ws/src/velodyne_simulator/velodyne_gazebo_plugins/src/GazeboRosVelodyneLaser.cpp:35:
/usr/include/gazebo-11/gazebo/msgs/test.pb.h: At global scope:
/usr/include/gazebo-11/gazebo/msgs/test.pb.h:187:41: error: ‘CachedSize’ in namespace ‘google::protobuf::internal’ does not name a type; did you mean ‘ToCachedSize’?
  187 |   mutable ::google::protobuf::internal::CachedSize _cached_size_;
      |                                         ^~~~~~~~~~
      |                                         ToCachedSize
/usr/include/gazebo-11/gazebo/msgs/subscribe.pb.h: In member function ‘virtual gazebo::msgs::Subscribe* gazebo::msgs::Subscribe::New() const’:
/usr/include/gazebo-11/gazebo/msgs/subscribe.pb.h:125:12: error: ‘CreateMaybeMessage’ was not declared in this scope
  125 |     return CreateMaybeMessage<Subscribe>(NULL);
      |            ^~~~~~~~~~~~~~~~~~
[... cut ...]
make[1]: *** [CMakeFiles/Makefile2:589: velodyne_simulator/velodyne_gazebo_plugins/CMakeFiles/gazebo_ros_velodyne_laser.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make[2]: *** [velodyne_simulator/velodyne_gazebo_plugins/CMakeFiles/gazebo_ros_velodyne_gpu_laser.dir/build.make:63: velodyne_simulator/velodyne_gazebo_plugins/CMakeFiles/gazebo_ros_velodyne_gpu_laser.dir/src/GazeboRosVelodyneLaser.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:562: velodyne_simulator/velodyne_gazebo_plugins/CMakeFiles/gazebo_ros_velodyne_gpu_laser.dir/all] Error 2
make: *** [Makefile:141: all] Error 2

reiniscimurs, 2024-01-06:
Hi,

These are catkin issues and unfortunately I cannot really help you with this. What you can try is to clone the repo again and use command `catkin_make` instead of `catkin_make_isolated`.

sympetrumz, 2024-01-06:
ive done the catkin_make too and seems to have the same error. do i need to create new workspace and clone into it?

reiniscimurs, 2024-01-06:
You should not do `catkin_make` after `catkin_make_isolated` and vice versa. Delete the cloned repo, clone a new one and try the other method. That is the only thing I can suggest.

closed 2024-04-28 by reiniscimurs

mob852, 2024-07-15:
i think it is a problem caused by the velodyne_simulator don't support gazebo11 it just support until gazebo9