reiniscimurs/DRL-robot-navigation #105: How to change the scanning range of a simulated LiDAR to 360 °

manxunyu, 2024-01-15:
Hello, thank you for your work. I referred to some questions raised by others about deploying to real vehicles. Regarding the scanning range of the LiDAR, I would like to ask how to change the scanning range of the LiDAR in the simulation (now -90 °~90 °) to the full range. Because the VLP16 lidar on my actual vehicle uses all 360 ° point clouds.

reiniscimurs, 2024-01-15:
Hi,

You should be able to do that by changing the plugin values: https://github.com/reiniscimurs/DRL-robot-navigation/blob/main/catkin_ws/src/multi_robot_scenario/xacro/p3dx/pioneer3dx.xacro#L27

manxunyu, 2024-01-15:
Perhaps I am blocking this setting can achieve the effect I want:
``min_ Angle="-1.57" max_ Angle="1.57"``

closed 2024-01-15 by manxunyu